dresden-elektronik/deconz-rest-plugin #1249: Xiaomi Aqara Wall Switch QBKG11LM Power Consuption

janjurek, 2019-02-12:
Hello, I've been poking around this for a long time now, but I was unsuccessful to get the "power" sensor data from the **Aqara Wall Switch**( `lumi.ctrl_ln1.aq1` - `QBKG11LM`). The cluster for power outlet is present in the deCONZ interface as you can see in image below, but the blank `Power 11` new sensor is only advertised in `pwa` calls but never added. I can clearly read the watts in the deCONZ, but I'm unable to retrieve them from the Rest API. Is there anything I can do?

[image]

ebaauw, 2019-02-12:
The REST API plugin expects endpoints 0x03 and 0x04 for the ZHAPower and ZHAConsumption sensors for the `lumi.ctrl_ln1.aq1`, same as for the `lumi.ctrl_ln2.aq1`.  I though they had the same Zigbee fingerprint, but this might be the first time I actually see a `ctrl_ln1`.  Funny that they keep the _On/Off Switch_ on endpoint 0x05 (in sync with the `ctrl_ln2`), but they moved the _Mains Power Outlet_ and 0x0053 endpoints "up" (endpoint 0x02 is the second output line on the `ctrl_ln2`).

Could you please list the REST API resources (lights and sensors) created for the switch?  I suspect the power is currently reported in the ZHAConsumption sensor.  Could you please confirm that the ZHASwitch sensor's state changes when you press the button?

janjurek, 2019-02-12:
Thank you for the quick reply and the insight into the endpoints. Bellow I listed all the light/sensors that the API sends. The first one `Bathroom Light` is the one that properly changes state when I toggle the switch. The `"ep": 2` is not present in the sensors at all. The `"ep": 3 (Power 11)` has similar output in`deCONZ` interface and shows what I believe might me the either some counter of used energy or consumption of the switch itself? (not sure here). I'm just attaching it here in case:

[image]

**API resources**
```
"2": {
    "etag": "868f9bb885e7263705bdba0d3f161af9",
    "hascolor": false,
    "manufacturername": "LUMI",
    "modelid": "lumi.ctrl_ln1.aq1",
    "name": "Bathroom Light",
    "state": {
      "alert": "none",
      "on": false,
      "reachable": true
    },
    "swversion": "06-25-2018",
    "type": "Smart plug",
    "uniqueid": "00:15:8d:00:02:a4:9d:d5-01"
  },
"11": {
    "config": {
      "on": true,
      "reachable": true,
      "temperature": 3200
    },
    "ep": 3,
    "etag": "0b8b69ba396deb93e3ae130bc98af2dc",
    "manufacturername": "LUMI",
    "modelid": "lumi.ctrl_ln1.aq1",
    "name": "Power 11",
    "state": {
      "lastupdated": "2019-02-12T17:55:20",
      "power": 0
    },
    "swversion": "06-25-2018",
    "type": "ZHAPower",
    "uniqueid": "00:15:8d:00:02:a4:9d:d5-03-000c"
  },
  "12": {
    "config": {
      "on": true,
      "reachable": true,
      "temperature": 3200
    },
    "ep": 5,
    "etag": "0b8b69ba396deb93e3ae130bc98af2dc",
    "manufacturername": "LUMI",
    "mode": 1,
    "modelid": "lumi.ctrl_ln1.aq1",
    "name": "Wall Bathroom",
    "state": {
      "buttonevent": 1002,
      "lastupdated": "2019-02-12T18:11:04"
    },
    "swversion": "06-25-2018",
    "type": "ZHASwitch",
    "uniqueid": "00:15:8d:00:02:a4:9d:d5-05-0012"
  }
```

ebaauw, 2019-02-12:
Endpoint 0x02 (which should have been exposed as ZHAPower) reports the current power in W; endpoint 0x03 (which should have been exposed as ZHAConsumption) reports the life-time consumption in kWh .  `state.power` is in W; `state.consumption` is in Wh, so the _Present Value_ from 0x03 is multiplied by 1000 to transform kWh to Wh.  The update of `/sensors/11` was probably when _Present Value_ was just under 0.5, so `state.power` reports 0.

PR should make the REST API use the right endpoints.

janjurek, 2019-02-12:
Perfect! I really appreciate how quickly you solved this issue, what a wonderful support!

janjurek, 2019-02-18:
I can verify that the issue has been resolved in the current release of plugin (`2.05.59`).

closed 2019-02-18 by janjurek

cobirnm, 2019-07-06:
Does end point updates automaticaly or do you have to press read? Mine only updates when I go to the rest api and press update.