dresden-elektronik/deconz-rest-plugin #1250: Looking to intergrate 73693 Sylvania Smart Bulbs

neuman1812, 2019-02-12:
Product: 
https://www.amazon.com/SYLVANIA-Changing-Equivalent-SmartThings-Assistant/dp/B0197840KQ

I'd like to provide the logs,  but I run deconz on a headless server, and have no access to the screen shots requested.   I have also not been able to find the logs output anywhere.  the command :

`: deCONZ --dbg-info=1 --dbg-aps=1 --dbg-zcl=1 --dbg-zdp=1 --http-port=80`

outputs error :
```
QXcbConnection: Could not connect to display 
Aborted
```

manup, 2019-02-12:
The light should already be supported, 2.05.58 did contain some fixes for SYLVANIA/LEDVANCE brand.
Did you reset the light while light search was running?

In a headless setup you need to add the minimal platform:

    deCONZ --dbg-info=1 --dbg-aps=1 --dbg-zcl=1 --dbg-zdp=1 --http-port=80 -platform minimal

neuman1812, 2019-02-12:
Current Version: 
Vendor dresden elektronik
Product ConBee
Version 2.05.58 / 1/30/2019
Firmware 262F0500

I removed the light from the Osram app first,  then when I put the app into search mode.  I turned of on, it flashed red/gree/blue  then turned it off.  That did not work so I tried other way,  put the app in seach,  then turned off the light, and back on where it flashed again.  

I can try again with the log now running.

neuman1812, 2019-02-12:
I tried it with the log running 

`deCONZ --dbg-info=1 --dbg-aps=1 --dbg-zcl=1 --dbg-zdp=1 --http-port=80 -platform minimal
libpng warning: iCCP: known incorrect sRGB profile
14:48:37:079 HTTP Server listen on address 0.0.0.0, port: 8080, root: /usr/share/deCONZ/webapp/
14:48:37:096 CTRL. 3.16.214:48:37:189 dev /dev/ttyAMA0 (/dev/ttyAMA0)
14:48:37:189 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
14:48:37:474 parent process sudo
14:48:37:474 gw run mode: normal
14:48:37:474 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
14:48:37:476 DB sqlite version 3.16.2
14:48:37:478 DB PRAGMA page_count: 30
14:48:37:478 DB PRAGMA page_size: 4096
14:48:37:478 DB PRAGMA freelist_count: 0
14:48:37:478 DB file size 122880 bytes, free pages 0
14:48:37:478 DB PRAGMA user_version: 6
14:48:37:478 DB cleanup
14:48:37:478 DB create temporary views
14:48:37:485 don't close database yet, keep open for 900 seconds
19:48:37:487 started websocket server at port 8088
19:48:37:491 discovery updated announce interval to 10 minutes
19:48:37:495 found node plugin: libde_rest_plugin.so - REST API Plugin
19:48:37:499 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
19:48:37:517 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
19:48:37:549 dev /dev/ttyAMA0 (/dev/ttyAMA0)
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
19:48:38:106 device state timeout ignored in state 2
19:48:39:156 device state timeout ignored in state 2
19:48:40:184 device state timeout ignored in state 2
19:48:41:183 device state timeout ignored in state 2
19:48:42:183 device state timeout ignored in state 2
19:48:42:718 dev /dev/ttyAMA0 (/dev/ttyAMA0)
19:48:42:719 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_Rpi_0x262f0500.bin.GCF
19:48:42:818 scan skip host .104
19:48:43:183 device state timeout (handled)
19:48:43:683 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
19:48:43:683 device state timeout ignored in state 1
19:48:43:875 Announced to internet
19:48:43:876 discovery server date: Tue, 12 Feb 2019 19:48:43 GMT
19:48:43:876 	 local time seems to be ok
19:48:43:876 discovery found version 2.04.35 for update channel stable
19:48:44:247 dev /dev/ttyAMA0 (/dev/ttyAMA0)
19:48:45:217 dev /dev/ttyAMA0 (/dev/ttyAMA0)
19:48:45:268 dev /dev/ttyAMA0 (/dev/ttyAMA0)
19:48:45:773 device state timeout ignored in state 2
19:48:46:723 device state timeout ignored in state 2
19:48:47:688 device state timeout ignored in state 2
19:48:48:683 device state timeout ignored in state 2
19:48:49:287 scan finished
19:48:49:683 device state timeout ignored in state 2
19:48:50:684 device state timeout (handled)
19:48:51:223 dev /dev/ttyAMA0 (/dev/ttyAMA0)
19:48:51:225 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
19:48:51:225 device state timeout ignored in state 2
19:48:51:703 device state timeout ignored in state 2
19:48:52:683 device state timeout ignored in state 2
19:48:53:683 device state timeout ignored in state 2
19:48:54:683 device state timeout ignored in state 2
19:48:55:684 device state timeout (handled)
19:48:56:220 dev /dev/ttyAMA0 (/dev/ttyAMA0)
19:48:56:222 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
19:48:56:222 device state timeout ignored in state 2
19:48:56:701 device state timeout ignored in state 2
19:48:57:684 device state timeout ignored in state 2
19:48:58:683 device state timeout ignored in state 2
19:48:59:684 device state timeout ignored in state 2
19:49:00:684 device state timeout (handled)
19:49:01:215 dev /dev/ttyAMA0 (/dev/ttyAMA0)
19:49:01:216 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
19:49:01:216 device state timeout ignored in state 2
19:49:01:701 device state timeout ignored in state 2
19:49:02:683 device state timeout ignored in state 2
`

It just repeats this device state line over and over...

manup, 2019-02-12:
This looks like the connection to RaspBee isn't working. Did you stop the deCONZ service before running the command?

    sudo systemctl stop deconz

Note I don't think debugging is needed here, the bulb should join normally.

To reset the light (while light search is running) you need to use the reset sequence:

https://www.youtube.com/watch?v=BKwvA_su9R0

manup, 2019-02-12:
Here is another reset sequence for SYLVANIA, not sure which one is the right one:

https://www.youtube.com/watch?v=KG1FWJBsa9g

neuman1812, 2019-02-12:
Rasbee?  Isn't that a different product?
https://www.dresden-elektronik.de/raspbee/

Im using a conbee stick.

I did not stop the deconz service,   I was unaware thats required.  I'll try the reset sequence.

manup, 2019-02-12:
>Rasbee? Isn't that a different product?
>https://www.dresden-elektronik.de/raspbee/
>Im using a conbee stick.

Ah ok, ConBee works as well thought it was a RaspBee.

neuman1812, 2019-02-12:
so... If I stop
 ```
sudo systemctl stop deconz
```

I cant access  the phoscan app?  How do  add a zigbee device?

manup, 2019-02-12:
Sorry my bad, stopping deCONZ service is only needed if the above debugging mode should run, since only one deCONZ instance is allowed to run.

So either start deCONZ with debugging enabled as mentioned above or just restart the service:

    sudo systemctl start deconz

neuman1812, 2019-02-12:
nope...still nothing.

manup, 2019-02-12:
Hmm hard to tell, does the light blink after reset as described in the reset video?

https://support.smartthings.com/hc/en-us/articles/213137623-How-to-reset-ZigBee-bulbs#Osram

neuman1812, 2019-02-12:
I did the reset according to the video https://www.youtube.com/watch?v=KG1FWJBsa9g   Since the bulb is a Smart+ .  I do get the green/red/blue flash.    According to that link,  its the same process.  5 on/off.

neuman1812, 2019-02-12:
If you have discord we can take this out of github if you want?  Im on the home assistant discord under the software room

neuman1812, 2019-02-12:
Im giving up on this right now.  Feel free to close this issue.  Im going to nuke the full install on deconz and restart from the beginning.

closed 2019-02-12 by neuman1812