dresden-elektronik/deconz-rest-plugin #1253: Using a contact (door/window) sensor

stevenwfoley, 2019-02-13:
I'm sorry if this has already been covered; I searched but could not find anything helpful.

I see that Phoscon supports adding contact sensors, but when editing a group and using the Sensor control feature I only see sensor types of motion and daylight. How are people actually using the contact sensors to control lights?

jnmarczi, 2019-02-13:
you have to add rules using the rest api... 

this is a sample rule of my setup. it switches a light group on, when the main entrance door opens. it will be fired from 6pm to 6am only.

` {
    "actions": [
      {
        "address": "/groups/39/action",
        "body": {
          "on": true,
          "transitiontime": 1
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/sensors/00:15:8d:00:02:88:cb:4e-01-0006/state/open",
        "operator": "eq",
        "value": "true"
      },
      {
      "address": "/config/localtime",
      "operator": "in",
      "value": "T18:00:00/T06:00:00"
			},
      {
        "address": "/sensors/00:15:8d:00:02:88:cb:4e-01-0006/state/lastupdated",
        "operator": "dx"
      }
    ],
    "name": "Switch lights on when door opens (from 6pm to 6am)"
  }
  `

stevenwfoley, 2019-02-15:
Thank you very much!
What are people using to POST and PUT?
The documentation mentions the Chrome app Postman, which did work, but I wondered if there's a defacto standard app that has been adopted for these tasks?

jnmarczi, 2019-02-15:
I'm prefering **Insomnia REST Client** [https://insomnia.rest/](url) over Postman... Had some issues running Postman on my machine.

closed 2019-06-22 by stale[bot]