dresden-elektronik/deconz-rest-plugin #1254: Dim group

halloamt, 2019-02-13:
Hi,

thanks for making deConz great :-)
Is there a possibility to send a dim up or dim down command to a group? That would make dimming a scene via API a lot more beautiful as not all lights would get the same level of brightness but birghtness stays relative to the original scene definition. I'd love to add that to the REST plugin but I cannot write C :-(

Greetings from the Black Forest

manup, 2019-02-13:
Thanks :)

It's not documented but you can send a `action.bri_inc` to a group.
Range is `-254 .. 254`

    PUT /api/<apikey>/groups/<id>/action
    
    { "bri_inc": 25 }

halloamt, 2019-02-13:
Perfect, thanks :-)
Another PR for iobroker.deconz is coming soon...

closed 2019-02-13 by halloamt