dresden-elektronik/deconz-rest-plugin #1255: CPU gradually increasing

petergridge, 2019-02-14:
I am running a raspbee on Home Assistant/hassio with deconz addon

Version 2.05.58
firmware 262F0500

Since installing the CPU has gradually increase about 6% per day until I restart deConz and the CPU reduces back to 6-7% starting point.

The PI eventually runs very hot and becomes unresponsive. 

Any advice on how to prevent this currently  automating a restart to deConz

manup, 2019-02-14:
The fix to this will be available in 2.05.59 later today.

Related issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1190

peer69, 2019-02-15:
Havent had this issue before. It appeared with 2.05.59 though. Had to go back to 58. About 5 Minutes after starting deconz cpu load goes to 100% and it becomes unresponsive for both, the api and the web interface.

manup, 2019-02-15:
I don't get it :/ what's happening here? Can you start 2.05.59 with debugging enabled:

    deCONZ --dbg-info=2 --dbg-aps=1 --http-port=80

Also please make sure only one instance of deCONZ is running.

peer69, 2019-02-19:
I gave 59 another try today. The error did not occur. But I noticed another CPU load related error. 
The release notes state you can get into an editor when pressing Alt + Click the Advanced Icon in the gateway settings. 
For me the editor does not open. But it seems to trigger something, because CPU load of deCONZ process jumps up to 100%. Web Interface and GUI become unresponsive, after restarting the service everything seems to go back to normal. 

[image]

This is a log output from when this happened, quite a vast output which certainly is the reason for the cpu load:
[deconz_log.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2879298/deconz_log.zip)

manup, 2019-06-20:
The issue was resolved in recent versions.

closed 2019-06-20 by manup