dresden-elektronik/deconz-rest-plugin #1256: RGBGenie ZB-5001 integration request

gmitch64, 2019-02-14:
Could the RGBGenie remote be added in please.

https://rgbgenie.com/?product=rgbgenie-5-zone-remote-and-dimmer-zigbee

Here's the cluster info I think that's needed - if there's anything else that's needed, please let me know. Feels like it's a nice remote, and not overly expensive ($30 from Amazon). Supports Zigbee 3.0

Thanks

Graham
[image]
[image]
[image]

gmitch64, 2019-06-14:
Hi,
Is there any chance that this remote could be integrated? Is there additional info that's needed?

G

jbkuma, 2019-08-18:
This would definitely be a nice add as a readily available wireless scene controller

gmitch64, 2019-12-16:
I'd still like to see this controller added please.
Is there any more info that I need to submit for it?

G

gmitch64, 2020-01-27:
Does anyone know if any progress has been made on this? If there's info the devs need, please let me know, and I can get it sent.

Thanks.

G

gmitch64, 2020-05-12:
Not sure how to link this to another Issue, but

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2732

SwoopX, 2020-05-17:
@gmitch64 I could have a look. However, I need some further screenshots from the power configuration cluster (with attributes being read) and the node containing all the endpoints and clusters.

SwoopX, 2020-06-07:
@gmitch64 so is the device now exposed via REST API? I've understood that at least some bottons are yet non-functional.

gmitch64, 2020-06-08:
OK. I've had a poke with Postman, and I'm not seeing it in the API - I've checked under sensors (that's where it should be?). I also had a look at lights too, but it wasn't there (and I didn't expect it to be).

So, it looks like it's not being exposed.

SwoopX, 2020-06-09:
I'd have expected to find it via sensors. I'll have another look when I get some time...

gmitch64, 2020-07-02:
@SwoopX Have you had a chance to look at this yet?

gmitch64, 2020-07-31:
I'm just wondering if you've had a chance to look at this one yet?

Thanks.

rylonian, 2020-09-09:
I'm also trying to get this remote working with deCONZ / Home Assistant.  

I have downloaded the latest version, 2.05.80, on a Windows 10 PC for testing purposes and have tried adding this remote through the Phoscon App and the Web App.

-If I try to add the remote via the Web App under "Switches", the remote pairs and appears in deCONZ, but the Web App never finds the device.  
-If I try to add the remote via the Web App under "Sensors", the remote pairs and appears in deCONZ and the Web App says that is successfully added one sensor, but the remote does not appear under Sensor or Switches. 

I have also installed deCONZ, 2.05.79, with Home Assistant and it seems to be the same as described above for version 2.05.80.  Even though the remote doesn't appear in the Web App, an entity is created for it in Home Assistant named "RGBgenie ZB-5001 Battery Level".  However, the deconz_event never triggers for any button press on the remote.  

I apologize, I'm very new to deCONZ, so I have yet to find out where to see if deCONZ itself is picking up the button push.

Any info or help on getting this working is very appreciated!

Mimiix, 2020-09-09:
@rylonian That is normal behaviour: Phoscon (the web app) needs to add support too. It does work with the API, but just not in the WebApp yet.

For information, read: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-for-Dummies

rweise, 2020-09-13:
Sorry, that i opened a new request sometimes 2 eyes are not enough ;-) but here are my experience with this remote under Raspbee2 / deConz 2.05.80 and with IOBroker 

The Device is optical like iCasa or Sunricher FB only that Buttons S1 and S2 here labeled as Pair 5 I/O
Phoscon will connect it as a Sensor and it is shown in the deconz-GUI 
[image]
[image]
[image]
ioBroker shows 5 Sensors with empty button-events.

SwoopX, 2020-09-13:
What the heck... @rweise Do you have 2 different versions of the same remote? One starting with "ZGRC" and the "RGBGenie ZB-5001"? The latter will not work as this model has not yet been taken into account.

rweise, 2020-09-13:
I have all 3 . Because i can not understand that there is no real remote system for zigbee i buy a lot and hope that there will be support in near future. I had ordered the iCasa because they are in the list of supported devices. same with the RGBgenie. At first i start with many different systems, but now i use only deconz.
Thats why i test all remotes that are named as working with deconz.
I own the icasa, the sunricher and the RGBgenie. I own the tuya Heiman clone and the Neo Keyfob
It is nice to control all with a smartphone, but for the light i prefer a remote as a easy way. The Heiman  or Neo are planed as "Masterswitches" at the Keychain. The other are mainly for the lightcontrol.

And because this is off topic,  thanks to all of you at Dresden-elektronik for there great job.

SwoopX, 2020-09-13:
Unfortunately, including remotes is the most troublesome task, especially when done remotely. When you see what's going over the air with a sniffer, this really helps a lot and speeds things up. Otherwise, it's more like tapping in the dark or doing the work more or less based on assumptions. Compared to other devices, I'd say it's the most complex work.

Are the other remotes working now so far?

rweise, 2020-09-14:
At the moment the sunricher and the icasa works very well. The rest will not work. thats why i use 2 innr RCs but they are not really nice in a livingroom . And i use 2 Bitron FBs for the "mainswitch" but they are also not really nice. That`s why i search my luck with the Neo Keyfob or Heiman. And for real, if it helps and there is a adress, i will offer a Neo Keyfob a Tuya TS-0215 Heiman Clone and a RGBgenie ZB-5001 for free and never return :-)
Dresden is not so far away form Magdeburg :-)

SwoopX, 2020-09-14:
Ping me at our Discord server ;)

closed 2020-09-15 by manup