dresden-elektronik/deconz-rest-plugin #1257: Relay switch compatibility

mredz, 2019-02-14:
To the support team,
I'm super happy with the Conbee so far thanks. 
I have this switch:
https://www.ebay.com.au/itm/Zigbee-OnOff-Controller-Single-Channel-Smart-Switch-For-Amazon-Alexa-SmartThings/263804968596?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649
It is working great except for one confounding thing. It switches on and off then on and off then on and off again remaining in the state that it started in. The light in Phoscon switches on and off with the fluctuations accordingly. It is also coming up as a light and not a switch but that is not a huge issue. It seems like it is super close to being compatible? Is there anything I can do on my side to tweak it to work?
[image]

Smanar, 2019-02-14:
It works on your side ?
Another person is trying too, but the device still supplies power in off or on state > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1090

mredz, 2019-02-14:
Actually I’ve not tested something on the relay yet so it may be started and finishing in the on position rather than the other way round but I can definitely hear it clicking on and off. So the trigger is working it’s just occurring too many times.

mredz, 2019-02-15:
Ok just to clarify this. I have put a light on the relay. The device will switch anywhere from 3-6 times. Sometimes returning the unit to an on state and sometimes an off state. There doesn't seem to be any rhyme or reason to how many times it will fluctuate.

manup, 2019-02-16:
Closing this issue, please continue discussion in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1090

closed 2019-02-16 by manup