dresden-elektronik/deconz-rest-plugin #1258: Device support request: Tradfri repeater

szoller, 2019-02-14:
Hello,

this week, I bought two of IKEA Tradfri Repeaters (Tradfri Signalverstärker: https://www.ikea.com/de/de/catalog/products/10400408/ (german).

According to the manual, I have to pair them with the tradfri hub, so i guess I have to pair them with my ConBee, too.
Unfortunately, they do not show up in DeConz-Software so I cannot provide further information.

I already tried to do a factory reset (5sec reset button) also as trying to do press the button 1,2 or more than 10 seconds, no change.
The repeaters have been placed in different locations, from another room also as in <2cm distance to the conbee stick.

Thanks for your work

Sebastian

manup, 2019-02-14:
Note there is already an issue for the IKEA Tradfri repeater: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1095

szoller, 2019-02-14:
I've completely missed that, sorry

manup, 2019-02-14:
All good, just need to keep it a bit organized here :)

tumik, 2019-02-17:
I was able to get the Trådfri repeater to show up like this:

1. Power on the repeater (plug it to wall socket)
2. Open the network (I clicked "Add new lights" from Phoscon App)
3. Reset the repeater by pushing a pin into the pinhole and holding it for at least 5 seconds

Shortly after the Trådfri repeater got visible in the deCONZ app, but it does not get visible in Phoscon.

SeraphimSerapis, 2019-04-26:
Can we have support for the repeater in the app? I know this is not necessarily a plug / switch / sensor -- might be worth it adding either a section for pure repeaters or adding it as a dumb sensor, though.

wvuyk, 2019-05-29:
One controls a sensor with the REST API, what is there to control on a repeater? It would be a useless entry right?

SeraphimSerapis, 2019-05-29:
Well, a GET method could be interesting. You'd not need to manipulate anything.

themfinger, 2019-05-29:
It's not easy to add the Repeater to the mesh.
Resetting wouldn't add. Through the Phoscon also I had no success.

What I did was log in through VNC into the port exposed by the Docker
Container and create a state to "listen" in the DECONz graphical utility.
After that, the device (that was already visible but with a generic
number-name), changed to a different name and started connecting to the
rest of the nodes.

Nuno Liça

dklueh79, 2019-08-04:
@themfinger:
Could you please post a few screenshots, how to use the tradfri as real repeater?
Thx!

Digital-Thor, 2019-12-01:
> I was able to get the Trådfri repeater to show up like this:
>
> 1. Power on the repeater (plug it to wall socket)
> 2. Open the network (I clicked "Add new lights" from Phoscon App)
> 3. Reset the [...]

Thanks for the instructions @tumik. They worked for me.  

**Note to others when pairing:** 
The pin hole is very small.  I used the bunt end of a sewing pin and pressed it for 10 seconds.  When I released the pin, the repeater's LED blinked.  

[image]

I also noticed that the repeater showed up in the [Node-Red deconz node](https://github.com/andreypopov/node-red-contrib-deconz).

[image]

RezzZ, 2020-03-22:
anyone able to connect new lamps near the tradfri repeater (but out of range of the conbeeII)? I tried several times to connect a tradfri bulb but it will not pair with deconz (via the repeater). I moved it near the conbee II and the bulb was found but after moving it upstairs the distance to the conbee was too big. And the bulb doesn't seem to find a new route trough the repeater. I read somewhere that I had to repair the bulb via the repeater but it will not work...

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix