dresden-elektronik/deconz-rest-plugin #1260: Support for Smarthings GP-U999SJVLBAA (Samjin) Motion Sensor

vinmar1, 2019-02-14:
Having these sensors would be great many people could get rid of the Smartthings Hub
Smarthings GP-U999SJVLBAA (Samjin) Motion Sensor

manup, 2019-02-21:
Can you please provide some details about the sensors as described in the Wiki:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

vinmar1, 2019-02-21:
I'm way new at this not sure how they pulled the info... is this what you need though?

Samjin motion
zha.zha_samjin_motion_01033d1a
Samjin motion
binary_sensor.samjin_motion_01033d1a_1_1280
Samjin motion
sensor.samjin_motion_01033d1a_1_1026

*****************
Basic (Endpoint id: 1, Id: 0x0000, Type: in)

alarm_mask (id: 0x0013)
app_profile_version (id: 0x0008)
app_version (id: 0x0001)
date_code (id: 0x0006)
device_enabled (id: 0x0012)
disable_local_config (id: 0x0014)
hw_version (id: 0x0003)
location_desc (id: 0x0010)
manufacturer (id: 0x0004)
model (id: 0x0005)
physical_env (id: 0x0011)
power_source (id: 0x0007)
stack_version (id: 0x0002)
sw_build_id (id: 0x4000)
zcl_version (id: 0x0000)

******************
IasZone (Endpoint id: 1, Id: 0x0500, Type: in)

cie_addr (id: 0x0010)
current_zone_sensitivity_level (id: 0x0013)
num_zone_sensitivity_levels_supported (id: 0x0012)
zone_id (id: 0x0011)
zone_state (id: 0x0000)
zone_status (id: 0x0002)
zone_type (id: 0x0001)

******************
Identify (Endpoint id: 1, Id: 0x0003, Type: in)

identify_commission_state (id: 0x0001)
identify_time (id: 0x0000)

******************
Identify (Endpoint id: 1, Id: 0x0003, Type: out)

identify_commission_state (id: 0x0001)
identify_time (id: 0x0000)

******************
Ota (Endpoint id: 1, Id: 0x0019, Type: out)

current_file_version (id: 0x0002)
current_zigbee_stack_version (id: 0x0003)
downloaded_file_version (id: 0x0004)
downloaded_zigbee_stack_version (id: 0x0005)
file_offset (id: 0x0001)
image_stamp (id: 0x000a)
image_type_id (id: 0x0008)
image_upgrade_status (id: 0x0006)
manufacturer_id (id: 0x0007)
minimum_block_req_delay (id: 0x0009)
upgrade_server_id (id: 0x0000)

*****************
PollControl (Endpoint id: 1, Id: 0x0020, Type: in)

checkin_interval (id: 0x0000)
checkin_interval_min (id: 0x0004)
fast_poll_timeout (id: 0x0003)
fast_poll_timeout_max (id: 0x0006)
long_poll_interval (id: 0x0001)
long_poll_interval_min (id: 0x0005)
short_poll_interval (id: 0x0002)

******************
PowerConfiguration (Endpoint id: 1, Id: 0x0001, Type: in)

battery_2_a_hr_rating (id: 0x0052)
battery_2_alarm_mask (id: 0x0055)
battery_2_alarm_state (id: 0x005e)
battery_2_percent_min_thres (id: 0x005a)
battery_2_percent_thres1 (id: 0x005b)
battery_2_percent_thres2 (id: 0x005c)
battery_2_percent_thres3 (id: 0x005d)
battery_2_percentage_remaining (id: 0x0041)
battery_2_quantity (id: 0x0053)
battery_2_rated_voltage (id: 0x0054)
battery_2_size (id: 0x0051)
battery_2_volt (id: 0x0050)
battery_2_volt_min_thres (id: 0x0056)
battery_2_volt_thres1 (id: 0x0057)
battery_2_volt_thres2 (id: 0x0058)
battery_2_volt_thres3 (id: 0x0059)
battery_2_voltage (id: 0x0040)
battery_3_a_hr_rating (id: 0x0072)
battery_3_alarm_mask (id: 0x0075)
battery_3_alarm_state (id: 0x007e)
battery_3_percent_min_thres (id: 0x007a)
battery_3_percent_thres1 (id: 0x007b)
battery_3_percent_thres2 (id: 0x007c)
battery_3_percent_thres3 (id: 0x007d)
battery_3_percentage_remaining (id: 0x0061)
battery_3_quantity (id: 0x0073)
battery_3_rated_voltage (id: 0x0074)
battery_3_size (id: 0x0071)
battery_3_volt (id: 0x0070)
battery_3_volt_min_thres (id: 0x0076)
battery_3_volt_thres1 (id: 0x0077)
battery_3_volt_thres2 (id: 0x0078)
battery_3_volt_thres3 (id: 0x0079)
battery_3_voltage (id: 0x0060)
battery_a_hr_rating (id: 0x0032)
battery_alarm_mask (id: 0x0035)
battery_alarm_state (id: 0x003e)
battery_percent_min_thres (id: 0x003a)
battery_percent_thres1 (id: 0x003b)
battery_percent_thres2 (id: 0x003c)
battery_percent_thres3 (id: 0x003d)
battery_percentage_remaining (id: 0x0021)
battery_quantity (id: 0x0033)
battery_rated_voltage (id: 0x0034)
battery_size (id: 0x0031)
battery_volt (id: 0x0030)
battery_volt_min_thres (id: 0x0036)
battery_volt_thres1 (id: 0x0037)
battery_volt_thres2 (id: 0x0038)
battery_volt_thres3 (id: 0x0039)
battery_voltage (id: 0x0020)
mains_alarm_mask (id: 0x0010)
mains_frequency (id: 0x0001)
mains_volt_max_thres (id: 0x0012)
mains_volt_min_thres (id: 0x0011)
mains_voltage (id: 0x0000)
mains_voltage_dwell_trip_point (id: 0x0013)

********************
TemperatureMeasurement (Endpoint id: 1, Id: 0x0402, Type: in)

max_measured_value (id: 0x0002)
measured_value (id: 0x0000)
min_measured_value (id: 0x0001)
tolerance (id: 0x0003)

manup, 2019-02-21:
Looks like an IAS Zone motion + temperature sensor, guess it can be added easily.

Do you have a deCONZ GUI available to make a few Screenshots from basic cluster and Node Info Panel as shown in the Wiki?

What's missing is the model identifier and verification of the manufacturer code from node descriptor (likely 0x1241 from Samjin).

Perhaps it is in this snipped, but hard to tell:
```
Samjin motion
zha.zha_samjin_motion_01033d1a
Samjin motion
binary_sensor.samjin_motion_01033d1a_1_1280
Samjin motion
sensor.samjin_motion_01033d1a_1_1026
```

vinmar1, 2019-02-21:
I could not get any of these to connect to the Conbee at all so I am currently using them with a HUSBZB-1. 
I looked in hassio and cannot find any other info other than this which I am sure isn't what you need. Any thought on how to pull the info, I'm new but can follow instructions pretty well.

nwk 0x6224
ieee 28:6d:97:00:01:03:3d:1a
lqi 255
rssi -61
battery level 186

vinmar1, 2019-02-21:
I did poll these but does not look like what you need I will keep poking, going to try again to get the Conbee to recognize it...

model (id: 0x0005)  value returned, b'motion'
manufacturer (id: 0x0004) value returned, b'Samjin'

vinmar1, 2019-02-21:
closed by accident

dewet22, 2019-03-01:
I'm similarly interested in getting these to work. All clusters:

[image]

Basic cluster:

[image]

Temperature Measurement cluster:

[image]

Let me know if you need anything else!

dewet22, 2019-03-01:
Also, mine is model number GP-U999SJVLBEA which I suspect is just the EU variant. Not sure if that makes any difference.

GadgetAngel, 2019-04-01:
I would love to see the Samjin Motion sensor to the Dresden Elektronik ZigBee USB stick

GadgetAngel, 2019-04-01:
here are my files:  

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

crazifyngers, 2019-04-08:
let me know if you need any more information.  I also have one.

closed 2019-08-14 by stale[bot]

ebaauw, 2020-04-11:
Does this device support Tamper detection and low battery in the IAS Zone Status?  The code was inconsistent that `state.tampered` and `state.lowbattery` were not exposed on pairing, but were on restore from the database.  I now don't expose them for both situations, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2672#issuecomment-612491012.