dresden-elektronik/deconz-rest-plugin #1262: Double Press being registered on single press (Xiaomi WXKG01LM)

wlatic, 2019-02-14:
Have just updated to the latest deconz

Version 2.05.58 / 30/01/2019
Firmware 262F0500

and now I'm getting double clicks registered on single clicks of the button.

{"event_type":"deconz_event","event":{"id":"switch_2","event":1003}}
In addition to this two of the events are being registered each time.

ebaauw, 2019-02-14:
x003 is release (after hold, aka long release).  Typically it's preceded by a x001 (hold).  Double press is x004.  See https://github.com/dresden-elektronik/deconz-rest-plugin/blob/4c37242fe13753c066ff9cbc4bc8dc0c07c9e8ab/sensor.h#L22-L32.

wlatic, 2019-02-14:
Sorry.

So its registering a long hold, rather than a single press.

wlatic, 2019-02-15:
Looks like the newest released version of the software resolves this.

closed 2019-06-22 by stale[bot]