dresden-elektronik/deconz-rest-plugin #1264: [Device Support Request] Samsung SmartThings Multipurpose Sensor

stevenwfoley, 2019-02-15:
Hello,
I am requesting support for the Samsung SmartThings Multipurpose Sensor (2016 model) STS-MLT-250.
Please and thank you.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Terra2000, 2019-05-08:
Yes Please

manup, 2019-05-08:
Mine should arrive soon, so the full integration should be available within the next weeks.

grahambelli, 2019-05-09:
Integration would be amazing

luisadnsaavedra, 2019-05-27:
Any news on the progress, will this integration be generic enough to support the SmartThings Motion Sensor (STs-IRM-251) which includes temperature?

manup, 2019-05-27:
Yes, following sensors are supported with deCONZ 2.05.65:
* Samjin Multipurpose Sensor
* Samjin Motion Sensor
* Samjin Water Sensor

Related commits:  https://github.com/dresden-elektronik/deconz-rest-plugin/commit/83b9e9a0aed14dc6da62d917716d8dff01202d3a https://github.com/dresden-elektronik/deconz-rest-plugin/commit/d1f2295494fdfaaa6c2d8eb2fc84aaedba44ab0c https://github.com/dresden-elektronik/deconz-rest-plugin/commit/d81cd529be58d0acd3efe3840d23cc409dd58611, the release notes follow later today.

Terra2000, 2019-06-26:
Any change for STS-MLT-251 ?

manup, 2019-06-26:
>Any change for STS-MLT-251 ?

Isn't this the Samjin Multipurpose Sensor? It was added in 2.05.65.

hellcry37, 2019-07-10:
I live in eu so i hve eu part of samjin_multipurpose_sensor that works ok, the issue i have with it is the part that does vibration/motion triggers every 1h approximately, no idea why. In my home assistant history display show sensor triggering every 1h

srg74, 2019-08-20:
> Yes, following sensors are supported with deCONZ 2.05.65:
> 
> * Samjin Multipurpose Sensor
> * Samjin Motion Sensor
> * Samjin Water Sensor
> 
> Related commits: [83b9e9a](https://github.com/dresden-elektronik/deconz-rest-plugin/commit/83b9e9a0aed14dc6da62d917716d8dff01202d3a) [d1f2295](https://github.com/dresden-elektronik/deconz-rest-plugin/commit/d1f2295494fdfaaa6c2d8eb2fc84aaedba44ab0c) [d81cd52](https://github.com/dresden-elektronik/deconz-rest-plugin/commit/d81cd529be58d0acd3efe3840d23cc409dd58611), the release notes follow later today.

At least Samjin Water Sensor and Samjin Multipurpose Sensor in my setup (Conbee II + Plug in for Home Assistant) always became unavailable over some time and working very not reliable. So in my "book" it's not fully supported. Sorry.

hellcry37, 2019-08-21:
> At least Samjin Water Sensor and Samjin Multipurpose Sensor in my setup (Conbee II + Plug in for Home Assistant deCONZ 2.05.66) always became unavailable over some time and working very not reliable. So in my "book" it's not fully supported. Sorry.

I use the same sensors that you say are unstable and have no disconnections, they are very reliable. I am most sure your disconnects and unavailability is due to wifi 2,4GHz interference. Read about this on the internet, I am sure yiu can solve the issue, is not related to deCONZ in any way.

srg74, 2019-08-21:
I don’t want to be rude.  But my statement  is not a wining. I have over 40 devices from 4 manufacturers which connected to the same Conbee 2 and all of them working just fine. Only Samsung(samjin) sensors is acting strange. So my conclusion is problem within those devices. And it’s have nothing to do with 2.4Ghz bandwidth because even here my wi-fi and zigbee channels is not nearby each other.

srg74, 2019-08-21:
P.S. sensors is very good working with Samsung hub in the same environment.

hellcry37, 2019-08-21:
@srg74 maybe that's the difference I don't have Samsung Smarthub anymore at all, maybe it interferes. Just saying I don't have any problem with them like disconnections, same setup HA / nuc8 / linux / docker and ConBee II

coding-columbo, 2019-08-28:
I would like to add my problem here too, I am unable to pair the samsung multipurpose sensor, it detects it and says ready, however when looking at the list of sensors it does not appear

srg74, 2019-08-30:
Deconz 2.05.67 - Samjin Multipurpose Sensor and Samjin Water Sensor disappears from sensor listing in Phoscon but remain visible in VNC. Is that a feature or a bug?

hellcry37, 2019-08-30:
same happens to me on deconz 2.05.67 - Samjin Multipurpose Sensor + that vnc is useless to be honest i cant delete it from there i cant do anything except view some info

hellcry37, 2019-09-10:
issue solved in Deconz 2.05.69
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1801

timn, 2019-09-19:
Just tried with a STS-MLT-251 (Samsung SmartThings Multipurpose Sensor), but it cannot pair:
```
21:30:49:290 APS-DATA.indication srcAddr: 0x3a4f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -22
21:30:49:291 ZCL got data for node=0x3A4F, cl=0x0500, at=0x0001, status=0x00, type=0x31
21:30:49:291 FP indication 0x0104 / 0x0500 (0x24FD5B00010BECBC / 0x3A4F)
21:30:49:291                       ...     (0x24FD5B00010BECBC / 0x3A4F)
21:30:49:291 Clear fast probe timeout for cluster 0x0500, 0x24FD5B00010BECBC
21:30:49:297 [3.1] get IAS Zone type for 0x24fd5b00010becbc
21:30:49:297 APS-DATA.request id: 40, addrmode: 0x02, addr: 0x3a4f, profile: 0x0104, cluster: 0x0500, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
21:30:49:539 CTRL skip polling while permit join is set
21:30:49:543 APS-DATA.confirm id: 40, status: 0x00 SUCCESS
21:30:49:543 APS-DATA.confirm request id: 40 -> erase from queue
21:30:49:544 APS-DATA.confirm id: 0, status: 0x00 SUCCESS
21:30:49:544 APS-DATA.confirm id: 0, status: 0x00, match: 0
21:30:49:562 skip device discovery while end devices is added
21:30:49:562 aps request id: 40 finished, erase from queue
```
Repeated for some time after pressing the connect button on the sensor. This is with a ConBee II and deconz 2.05.69 running in Docker. An Aqara sensor paired just fine.

I ran this with the VNC GUI and it seems that it has actually been paired, but it doesn't show in the web app and does not have a name.

Is there a trick to get this working or can I do more to help debugging?

townsmcp, 2019-09-21:
Same for me too with STS-MLT-251 - can add in VNC GUI but not in Phoscon app

jjanderson, 2019-09-23:
it would be great if the same devices that are in VNC can be viewed and managed in the Web UI... I have close on 70 Nodes.... My Smartthings Motion sensors keep failing (I hate them actually and have moved on) but I have 4 of them (2015 model) and still want to use them.

trying to find them in the VNC UI is a real pain.... it would be good if they could show up in the UI.. .The same goes for the Xiaomi smoke sensors....

sirpug, 2019-10-23:
ConBee II, HassOS 3.5, HA 0.100.3, deCONZ 3.6 addon, gateway version 2.05.69. Same issue as @townsmcp, @timn, and @stevenwfoley, but with STS-IRM-250 (motionv5 sensor) and STS-MLT-250 (multiv4 sensor). After going through "Add new Sensor" > "Other" it fails with the message "Failed to connect." Logging in to VNC shows both sensors in network and cluster info can be read.

sicutero, 2019-10-25:
Exactly same setup as @sirpug with the STS-MLT-250 sensor.

jhays, 2019-11-26:
I am running deconz rest api 2.05.71, 

I cannot get my SmartThings STS-MLT-250 sensor to pair. 

I am trying to add a sensor through the Phoscon app web interface, I start the scan by adding a device and selecting "other" brand, reset the sensor and place it close to my raspberry pi/conbee II, and nothing happens. 

I've also tried a second STS-MLT-250, and a SmartThings motion sensor. So far nothing will pair successfully.

Can anyone offer any tips on how to pair these devices? 

Thanks.

NearlCrews, 2020-03-13:
Any chance of getting the STS-MLT-250 added? I'm running 2.05.75 and have the same issue as the original poster. I've got 6 of them attached to SmartThings that I'd love to migrate over to deCONZ.

RichJeanes, 2020-05-12:
@manup  @ebaauw 
Is there any progress on getting the STS-MLT-250 integrated? If there's anything I can do or provide to aide the effort, please let me know. I just picked up a ConBee II and would like to migrate over from the SmartThings Hub.

ebaauw, 2020-05-12:
See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1848

RichJeanes, 2020-05-12:
I'm not sure now. The STS-MLT-250 is apparently a Sam Jin device (https://www.cleancss.com/user-manuals/2AF4S/-STS-MLT-250) but it does have a different outward appearance to the GP-U999SJVLAEA.

Though my issue with the device not pairing may be unrelated. I was able to pair my SmartThings motion sensor with my ConBee II plugged into my Windows computer running Phoscon, but it will not pair anything plugged into my Synology NAS running Home Assistant with the deCONZ addon on a virtual machine.

RichJeanes, 2020-05-12:
I was not able to add the STS-MLT-250 to the Phoscon app running on my Windows machine, so I think they may not be the same device internally.

RichJeanes, 2020-05-12:
Connecting the sensor in the deCONZ app in Windows, I can see that it has the Sam jin cluster 0xfc02 (same as the initiator of this issue), so it might be Sam Jin, but different internals (as stated in [#1311](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1311#issue-416204555)).

[image]

ebaauw, 2020-05-12:
The code should check on `modelId.startsWith(QLatin1String(‘multi’))` matching both models (assuming the other parameters are unchanged, _Manufacturer Code_, and mac address prefix.

RichJeanes, 2020-05-12:
ManufacturerName is 'SmartThings', not 'Samjin'. This difference is also documented in Samsung's [Groovy code](https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/master/devicetypes/smartthings/smartsense-multi-sensor.src/smartsense-multi-sensor.groovy#L35) for the devices, which was referenced when adding the Sam Jin model.
[image]

Per de_web_plugin.cpp, the mac prefix matches stMacPrefix, not samjinMacPrefix
[image]
[image]

markmghali, 2020-05-13:
I am also having this issue with the Smartthings 2016 Multi-purpose sensor
I can pair it through Phoscon it does not show up in the webui, but it shows in VNC.

I get temp and open/close but not Vibration.

I would really like vibration to work
let me know if I can assist in troubleshooting 
running deconz as a addon in home assistant 

[image]
[image]

ebaauw, 2020-05-13:
What _Manufacturer Code_ does it show in the _Node Descriptor_?

RichJeanes, 2020-05-13:
Sorry, new to this. Didn't realize there were both Code and Name.
Manufacturer Code = 0x110a

All of the info on my sensor has been the same as the screenshots at the top of this Issue.

markmghali, 2020-05-13:
[image]

RichJeanes, 2020-05-13:
@markmghali It seems like you may have a different model. Does it say "STS-MLT-250" on the back? That is the sensor this Issue was created for.

markmghali, 2020-05-13:
Oh, I might be in the wrong place?

ebaauw, 2020-05-13:
> The code should check on modelId.startsWith(QLatin1String(‘multi’)) matching both models

Damn.  It doesn't.

> I get temp and open/close but not Vibration.

@markmghali Could you please list the REST API resources for temperature and open/close?  I think you have a Centralite 3321-S?  Does that even support vibration?  Could you please add a screenshot of the node and the clusters?

So there seem to be three different models:

Mac Prefix | MF Code | Mf Name | Model Name  | Who has it?
---- | --------- | --------- | ------------- | -----
`0x000d6f` | `0x104e` |  ? | `3321-S` | @markmghali 
`0x24fd5b` |` 0x110a` | `SmartThings` | `multiv4` | @stevenwfoley @RichJeanes 
`0x286d97` | `0x1241` | `Samjin` | `multi` | @ebaauw

Apart from these attributes, there are other subtle differences: the `multiv4` model doesn't report _Battery Percentage Remaining_, only _Battery Voltage_.

The real question is what _Manufacturer Code_ the `multiv4` model uses for the 0xfc02 cluster.

@RichJeanes Could you please download `general.xml` from my latest commit, install it to `/usr/share/deCONZ/zcl`, restart deCONZ and check in the GUI that the cluster is now populated, and the vibration attributes can be read?

RichJeanes, 2020-05-13:
@ebaauw  The Manufacturer Name for the Centralite 3320, 3321, and3321-S should be "CentraLite" per [Samsung's Groovy Code](https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/master/devicetypes/smartthings/smartsense-multi-sensor.src/smartsense-multi-sensor.groovy#L32) for the sensors.

RE: `general.xml` - On Windows, I assume that be replacing the `general.xml` at `C:\Users\rich\AppData\Local\deCONZ\zcl`?

ebaauw, 2020-05-13:
I wouldn't know, I don't use Windows.  It's replacing the existing `general.xml` alright.

markmghali, 2020-05-13:
Where do I find the REST API resources for the Temp and Open/Close Sensors?

I do know it has vibration as it was working in Smartthings hub and in Home assistant through Smartthings Hub integration

I screenshotted what I could find:
 
[image]

[image]

[image]

[image]

RichJeanes, 2020-05-13:
The cluster is populated, but I don't get any values when I click `read`.
[image]

ebaauw, 2020-05-13:
@RichJeanes Did you wake the sensor when pressing _Read_?

RichJeanes, 2020-05-13:
I was shaking it while touching and removing the contact sensor magnet (the colored dot on the object tile blinks green when doing so).

ebaauw, 2020-05-13:
Could you try again, please, with latest version.  Could you try and set _Motion Threshold Multiplier_ to 1 and _Motion Threshold_ to 630 (values from the `.groovy` file)?

Note that the sensor actually reports _Active_ (at least the `multi` does), so resetting `state.presence` based on `config.duration` isn't needed at all.  I sure wish I'd known that earlier.

ebaauw, 2020-05-13:
@markmghali , could you also please try the `general.xml` and see if you can read the Acceleration attributes?

RichJeanes, 2020-05-13:
I was able to write the Multiplier, but the threshold fails to write. Attempting a read returns "unsupported attribute". Should the attribute id be `0x0002`?
From `.groovy`:
`configCmds += zigbee.writeAttribute(0xFC02, 0x0002, 0x21, "7602", [mfgCode: manufacturerCode])`

ebaauw, 2020-05-13:
> Should the attribute id be 0x0002?

Yeah, of course.

> For some reason, the read button on the Cluster Info pane does not seem to work for this cluster.

The GUI doesn't like to read manufacturer-specific attributes.

markmghali, 2020-05-14:
> @markmghali , could you also please try the `general.xml` and see if you can read the Acceleration attributes?

I need a little more instruction on how to try the general.xml? I am new to all this my apologizes.

RichJeanes, 2020-05-14:
@markmghali 
1. Go to (Windows) C:\users\[your user]\AppData\Local\deCONZ\zcl or (Linux) /usr/share/deCONZ/zcl
and rename `general.xml` to `general.xml.bak`
1. Open [this version](https://raw.githubusercontent.com/ebaauw/deconz-rest-plugin/master/general.xml) of `general.xml` and save it in `.../deCONZ/zcl` folder
1. Launch deCONZ and see if the 0xFC02 cluster of the sensor has attributes in it or if it is still a blank panel.

markmghali, 2020-05-14:
@RichJeanes 
Thank you for your reply

I am having a heck of time trying to figure out how to get to the deconz command line or using winscp.

I am running deconz as an add-on in Home Assistant. So running Home Assistant as a VM and deconz is a docker inside of Home Assistant, is my understanding. Googling is not yielding helpful results. 

Any assistance on this piece? I can keep poking around too. Excuse my lack of knowledge

RichJeanes, 2020-05-14:
@markmghali Then I think you want the `/usr/share/deCONZ/zcl/` directory in your deCONZ container. I think the easiest way might be to get a local shell on the deCONZ container.
1. In the HA VM, `docker exec -it [container name] bash`
1. Use curl or wget to pull the `general.xml` file directly from [this URL](https://raw.githubusercontent.com/ebaauw/deconz-rest-plugin/master/general.xml) into the `zcl` directory (see above about renaming the existing `general.xml` just in case).
1. `exit`
1. Restart the deCONZ container, either directly from the HA VM command line or in HA (Supervisor -> deCONZ -> Restart Add-on

As a side note, that's the same way I'd like to run it, but for some reason it looks like the deCONZ backend and Phoscon frontend being displayed in HA aren't communicating. It says it can't even connect to the virtual Daylight sensor. That's why I've been doing my debugging on my Windows desktop with deCONZ running locally.

markmghali, 2020-05-14:
I tried the docker exec and it does not work? I think this is using different commands? start with HA
Also I looked in /usr/share/ in Home Assistant and do not see a deCONZ folder?

I am guessing I am totally doing something wrong.
  
[image]

RichJeanes, 2020-05-14:
You're in the HA shell. You need to type `login` to drop to a full local shell. I didn't notice the instruction at first either, but it tells you that right below the login banner.

markmghali, 2020-05-14:
Ok Duh thank you 

after docker ps 
what is the proper name to use?
[image]

markmghali, 2020-05-14:
Ok I renamed general.xml to general.xml.oldbackup

I tried running curl and it pulled it down? but didn't create a general.xml file?

Again my lack of knowledge. Thank you for bearing with me.

[image]

markmghali, 2020-05-14:
Ok 0xFC02 still shows blank?
[image]

ebaauw, 2020-05-14:
Did you restart deCONZ after downloading the general.xml file?

markmghali, 2020-05-14:
Yes I restarted deconz and then I restarted all of home assistant.

RichJeanes, 2020-05-14:
@markmghali With curl you need to specify `-o <output file>`. It defaults to STDOUT.

[8 more replies]

closed 2020-07-25 by stale[bot]