dresden-elektronik/deconz-rest-plugin #1266: deCONZ Homebridge Service no longer starts with v2.05.59

saurabh984, 2019-02-15:
deconz homebridge integration doesn't start or show any logs after upgrading to v2.05.59.

How do I get more debug logs?

manup, 2019-02-15:
Ah sorry we did forgot the latest commit. I've updated the Raspbian package:

https://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.05.59-qt5.deb

Guess the service needs to be restarted afterwards.

To debug the homebridge script you can raise the debug level in `/usr/bin/deCONZ-homebridge.sh`

saurabh984, 2019-02-15:
Still no luck

`Feb 15 16:40:59 deconz deCONZ-homebridge.sh[10876]: /usr/bin/deCONZ-homebridge.sh: line 244: syntax error near unexpected token `done'
Feb 15 16:40:59 deconz deCONZ-homebridge.sh[10876]: /usr/bin/deCONZ-homebridge.sh: line 244: `#011#011#011done'
F`

saurabh984, 2019-02-15:
Changed the file `/usr/bin/deCONZ-homebridge.sh` with latest in git fixed it!

closed 2019-02-15 by saurabh984

manup, 2019-02-15:
Yes commit is (should be) also in the updated package :)