dresden-elektronik/deconz-rest-plugin #1267: Changed my username on the raspberry host (that has Conbee) now deCONZ does not connect

Klagio, 2019-02-15:
Now I can't connect to my Conbee anymore. When from terminal I launch deCONZ before it would launch the GUI and connect immediately to the Conbee, but not it stayus NOT CONNECTED, and not sure what I have to do

manup, 2019-02-15:
You need to add the new user to the group dialout so that it has access to the usb device:

    sudo gpasswd -a <user> dialout

Klagio, 2019-02-15:
```

pi@pioffice:~ $ deCONZ --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=1
libEGL warning: DRI2: failed to authenticate
libpng warning: iCCP: known incorrect sRGB profile
16:01:06:255 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
16:01:06:274 CTRL. 3.16.216:01:06:532 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
16:01:06:532 dev /dev/ttyAMA0
16:01:06:532 COM: /dev/ttyUSB0 : FT230X Basic UART (0x0403/0x6015)
16:01:06:533 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
16:01:06:866 parent process /bin/bash
16:01:06:866 gw run mode: normal
16:01:06:866 GW sd-card image version file does not exist: /home/pi/.local/share/dresden-elektronik/deCONZ/gw-version
16:01:06:866 sd-card cid: 035344534c31364780ffffffff010813
16:01:06:868 DB sqlite version 3.16.2
16:01:06:870 DB PRAGMA page_count: 30
16:01:06:870 DB PRAGMA page_size: 4096
16:01:06:871 DB PRAGMA freelist_count: 0
16:01:06:871 DB file size 122880 bytes, free pages 0
16:01:06:871 DB PRAGMA user_version: 6
16:01:06:871 DB cleanup
16:01:06:872 DB create temporary views
16:01:06:873 DB view [0] created
16:01:06:874 DB view [1] created
16:01:06:874 DB view [2] created
16:01:06:875 DB view [3] created
16:01:06:875 sql exec SELECT apikey,devicetype,createdate,lastusedate,useragent FROM auth
16:01:06:875 sql exec SELECT key FROM config2
16:01:06:875 sql exec SELECT key,value FROM config2
16:01:06:876 Load config UTC: 2019-02-15T15:44:33 from db.
16:01:06:876 Load config announceinterval: 10 from db.
16:01:06:876 Load config announceurl: http://dresden-light.appspot.com/discover from db.
16:01:06:876 Load config apiversion: 2.05.58 from db.
16:01:06:876 Load config bridgeid: 0000000000000000 from db.
16:01:06:876 Load config datastoreversion: 60 from db.
16:01:06:876 Load config dhcp: true from db.
16:01:06:876 Load config discovery: false from db.
16:01:06:876 Load config factorynew: false from db.
16:01:06:876 Load config fwneedupdate: false from db.
16:01:06:876 Load config fwupdatestate: idle from db.
16:01:06:876 Load config fwversion: 0x00000000 from db.
16:01:06:876 Load config gateway: 127.0.0.1 from db.
16:01:06:876 Load config groupdelay: 50 from db.
16:01:06:876 Load config gwpassword: $1$8282jdkm$w.o6XjJEGLzAYkVqLPAYx1 from db.
16:01:06:876 Load config gwusername: delight from db.
16:01:06:876 Load config homebridge: not-managed from db.
16:01:06:877 Load config homebridge-pin:  from db.
16:01:06:877 Load config homebridgepin:  from db.
16:01:06:877 Load config ipaddress: 192.168.1.238 from db.
16:01:06:877 Load config linkbutton: false from db.
16:01:06:877 Load config localtime: 2019-02-15T15:44:33 from db.
16:01:06:877 Load config mac: 38:60:77:7c:53:18 from db.
16:01:06:877 Load config modelid: deCONZ from db.
16:01:06:877 Load config name: Phoscon-GW from db.
16:01:06:877 Load config netmask: 255.0.0.0 from db.
16:01:06:877 Load config networkopenduration: 60 from db.
16:01:06:877 Load config otauactive: false from db.
16:01:06:877 Load config otaustate: off from db.
16:01:06:877 Load config panid: 0 from db.
16:01:06:877 Load config permitjoin: 0 from db.
16:01:06:878 Load config permitjoinfull: 0 from db.
16:01:06:878 Load config port: 8080 from db.
16:01:06:878 Load config portalservices: false from db.
16:01:06:878 Load config proxyaddress: none from db.
16:01:06:878 Load config proxyport: 0 from db.
16:01:06:878 Load config rfconnect: 1 from db.
16:01:06:878 Load config rfconnected: false from db.
16:01:06:878 Load config rgbwdisplay: 1 from db.
16:01:06:878 Load config runmode: normal from db.
16:01:06:878 Load config starterkitid:  from db.
[... 629 lines cut ...]
16:01:18:385 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
16:01:18:385 dev /dev/ttyAMA0
16:01:18:385 COM: /dev/ttyUSB0 : FT230X Basic UART (0x0403/0x6015)
16:01:18:453 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
16:01:18:453 dev /dev/ttyAMA0
16:01:18:453 COM: /dev/ttyUSB0 : FT230X Basic UART (0x0403/0x6015)
16:01:18:463 auto connect com /dev/ttyAMA0
16:01:18:471 Serial com disconnected, reason: 4
^C16:01:18:875 shutdown after signal(2)
16:01:18:905 saved node state in 0 ms
16:01:18:957 sync() in 51 ms
16:01:18:958 REST API plugin shutting down
16:01:18:958 DB save zll database items 0x000000C1
16:01:18:959 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode) VALUES ('1', 'Daylight', 'Daylight', 'PHDL00', 'Philips', '00:00:00:00:00:00:00:00-01', '1.0', '{"dark":false,"daylight":true,"lastupdated":"2019-02-15T16:01:06","status":170}', '{"configured":true,"lat":"41.902783","long":"12.496365","on":true,"sunriseoffset":30,"sunsetoffset":-30}', '', 'normal', '1')
16:01:18:977 DB saved in 18 ms
16:01:18:997 sync() in 18 ms
16:01:19:001 http client 0x0x1a0d180 unregister client handlers
16:01:19:001 ~Resource() /sensors 0x19c4c78
16:01:19:002 ~Resource() /groups 0x1a1adb0
16:01:19:002 ~Resource() /config 0x19a6714
16:01:19:545 ~http client 0x0x1a0d180

```

> sudo gpasswd -a <user> dialout

Klagio, 2019-02-15:
> You need to add the new user to the group dialout so that it has access to the usb device:
> 
> ```
> sudo gpasswd -a <user> dialout
> ```

Thanks I did this, but nothing same problem (sorry for the long log).

Klagio, 2019-02-15:
the new log

```

:09:17:944 Serial com disconnected, reason: 4
17:09:18:839 Daylight now: goldenHour2, status: 180, daylight: 1, dark: 0
17:09:18:911 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
17:09:18:911 dev /dev/ttyAMA0
17:09:18:912 COM: /dev/ttyUSB0 : FT230X Basic UART (0x0403/0x6015)
17:09:18:979 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
17:09:18:980 dev /dev/ttyAMA0
17:09:18:980 COM: /dev/ttyUSB0 : FT230X Basic UART (0x0403/0x6015)
17:09:18:993 auto connect com /dev/ttyUSB0
17:09:18:000 Serial com disconnected, reason: 4
17:09:19:903 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
17:09:19:903 dev /dev/ttyAMA0
17:09:19:903 COM: /dev/ttyUSB0 : FT230X Basic UART (0x0403/0x6015)
17:09:19:973 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
17:09:19:973 dev /dev/ttyAMA0
17:09:19:973 COM: /dev/ttyUSB0 : FT230X Basic UART (0x0403/0x6015)
17:09:19:986 auto connect com /dev/ttyAMA0
17:09:19:989 Serial com disconnected, reason: 4
17:09:20:903 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
17:09:20:903 dev /dev/ttyAMA0
17:09:20:903 COM: /dev/ttyUSB0 : FT230X Basic UART (0x0403/0x6015)
17:09:20:959 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
17:09:20:959 dev /dev/ttyAMA0
17:09:20:959 COM: /dev/ttyUSB0 : FT230X Basic UART (0x0403/0x6015)
17:09:20:966 auto connect com /dev/ttyUSB0
17:09:20:970 Serial com disconnected, reason: 4
17:09:21:904 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
17:09:21:905 dev /dev/ttyAMA0
17:09:21:905 COM: /dev/ttyUSB0 : FT230X Basic UART (0x0403/0x6015)
17:09:21:975 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
17:09:21:975 dev /dev/ttyAMA0
17:09:21:975 COM: /dev/ttyUSB0 : FT230X Basic UART (0x0403/0x6015)
17:09:21:995 auto connect com /dev/ttyAMA0
17:09:22:002 Serial com disconnected, reason: 4
17:09:22:905 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
17:09:22:905 dev /dev/ttyAMA0
17:09:22:905 COM: /dev/ttyUSB0 : FT230X Basic UART (0x0403/0x6015)
17:09:22:973 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
17:09:22:973 dev /dev/ttyAMA0
17:09:22:973 COM: /dev/ttyUSB0 : FT230X Basic UART (0x0403/0x6015)
17:09:22:984 auto connect com /dev/ttyUSB0
17:09:22:985 M-SEARCH * HTTP/1.1
Host: 239.255.255.250:1900
Man: "ssdp:discover"
ST: ssdp:all
MX: 5

17:09:22:987 M-SEARCH * HTTP/1.1
Host: 239.255.255.250:1900
Man: "ssdp:discover"
ST: ssdp:all
MX: 5

17:09:22:988 M-SEARCH * HTTP/1.1
Host: 239.255.255.250:1900
Man: "ssdp:discover"
ST: ssdp:all
MX: 5

17:09:22:993 Serial com disconnected, reason: 4
17:09:23:902 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
17:09:23:902 dev /dev/ttyAMA0
17:09:23:902 COM: /dev/ttyUSB0 : FT230X Basic UART (0x0403/0x6015)
17:09:23:971 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
17:09:23:971 dev /dev/ttyAMA0
17:09:23:971 COM: /dev/ttyUSB0 : FT230X Basic UART (0x0403/0x6015)
17:09:23:981 auto connect com /dev/ttyAMA0
17:09:23:984 Serial com disconnected, reason: 4
^C17:09:24:315 shutdown after signal(2)
17:09:24:344 saved node state in 0 ms
17:09:24:400 sync() in 55 ms
17:09:24:400 REST API plugin shutting down
17:09:24:401 DB save zll database items 0x000000C1
17:09:24:402 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode) VALUES ('1', 'Daylight', 'Daylight', 'PHDL00', 'Philips', '00:00:00:00:00:00:00:00-01', '1.0', '{"dark":false,"daylight":true,"lastupdated":"2019-02-15T16:08:08","status":180}', '{"configured":true,"lat":"41.902783","long":"12.496365","on":true,"sunriseoffset":30,"sunsetoffset":-30}', '', 'normal', '1')
17:09:24:415 DB saved in 14 ms
17:09:24:428 sync() in 11 ms
17:09:24:433 ~Resource() /sensors 0x9008f8
17:09:24:433 ~Resource() /groups 0x801fe8
17:09:24:433 ~Resource() /config 0x8e2b8c

```

manup, 2019-02-15:
Did you also restart, or logout + login?

Klagio, 2019-02-15:
> Did you also restart, or logout + login?

YEs after a reboot now it connects ... but I lost all my nodes!!! 64+   ... HELP!!

Klagio, 2019-02-15:
Now I see that automatically it adds components (but all without names) and if I go on the phoscon app, its all lost, there is nothing inside .,.... ohh no!

Klagio, 2019-02-15:
Is there anything I can do to recover it, I do not have any backup, I just changed my host name and added a user

Klagio, 2019-02-15:
If I log with my previous user and launch deCONZ this is what I get

openhabian@pioffice:/home/pi $ deCONZ
No protocol specified
QXcbConnection: Could not connect to display :0.0
Aborted

manup, 2019-02-15:
You also need to copy the the folder `/home/pi/.local/share/dresden-elektronik` to the new user home directory. All the node information is stored there.

Klagio, 2019-02-15:
@manup Hi sorry to bug you, but house is dark and need my lights, back, yes was stupid from my side

Klagio, 2019-02-15:
> /home/pi/.local/share/dresden-elektroni
trying to understand where this directory is, can't find it

pi@pioffice:~ $ cd /home/pi/.local/share/dresden-elektroni
-bash: cd: /home/pi/.local/share/dresden-elektroni: No such file or directory
pi@pioffice:~ $

manup, 2019-02-15:
Note deCONZ should be closed first.

    cp -r /home/pi/.local/share/dresden-elektronik /home/<new user>/.local/share

Klagio, 2019-02-15:
> Note deCONZ should be closed first.
> 
> ```
> cp -r /home/pi/.local/share/dresden-elektronik /home/<new user>/.local/share
> ```

yes thanks a lot

closed 2019-02-15 by Klagio

manup, 2019-02-15:
And always make backups, just in case :)