dresden-elektronik/deconz-rest-plugin #1269: Each time I reboot my raspberry pi, I do need to type deCONZ to make it start, anyway to make it start automatically after reboot?

Klagio, 2019-02-15:
Sorry Linux newb here

Smanar, 2019-02-15:
On my side I have created a deCONZ.desktop file, with the same content than the /usr/share/applications/deCONZ.desktop file.

But don't work on headless system.

manup, 2019-02-15:
You need to enable the deconz.service (headless) or deconz-gui.service (gui)

https://github.com/dresden-elektronik/deconz-rest-plugin#headless-support

Klagio, 2019-02-15:
> Are you using headless mode ?
> On my side I have created a deCONZ.desktop file (in ~/.config/autostart), with the same content than the /usr/share/applications/deCONZ.desktop file.
> 
> But don't work on headless system.

no I use GUI

Klagio, 2019-02-15:
> You need to enable the deconz.service (headless) or deconz-gui.service (gui)
> 
> https://github.com/dresden-elektronik/deconz-rest-plugin#headless-support

so I have to do this?

sudo systemctl enable deconz-gui

manup, 2019-02-15:
Yes. But since you are running under a different user now you might need to edit the systemd service, since it will start deCONZ as user with id 1000 (which is pi).

    sudo systemctl edit deconz-gui

Add line:

    User=<username>

Klagio, 2019-02-15:
> Yes. But since you are running under a different user now you might need to edit the systemd service, since it will start deCONZ as user with id 1000 (which is pi).
> 
> ```
> sudo systemctl edit deconz-gui
> ```
> Add line:
> 
> ```
> User=<username>
> ```

Ok, just to make sure I understand, I did
sudo systemctl edit deconz-gui

but it comes an empty file. Is that correct? Then I will add 
User=pi

manup, 2019-02-16:
>Ok, just to make sure I understand, I did
sudo systemctl edit deconz-gui
but it comes an empty file. Is that correct? Then I will add
User=pi

Yep, that should work (but frankly I never testet this myself).

closed 2019-06-23 by stale[bot]

Klagio, 2019-11-15:
> Are you using headless mode ?
> On my side I have created a deCONZ.desktop file (in ~/.config/autostart), with the same content than the /usr/share/applications/deCONZ.desktop file.
> 
> But don't work on headless system.

Hi, I did this, and it works. Only pèroblem that it is needed to put credentials at start (user and password) makinng it not ideal since it still needs human interaction to make it start. 

YOu have same problem, or how did you solve?

Smanar, 2019-11-15:
This method is not the easier ^^.
But no, I don't need to type user and password. You are speaking for deconz or phoscon ? Deconz never ask for password.

Klagio, 2019-11-16:
> This method is not the easier ^^. The @manup one don't works for you ?
> But no, I don't need to type user and password. You are speaking for deconz or phoscon ? Deconz never ask for password, it s not your system that ask that ?

Debian

Smanar, 2019-11-16:
I think it's from your system configuration and I don't know Unix at all.
Try just with enabling the service

`$ sudo systemctl enable deconz-gui`

mmarras, 2021-10-05:
> You need to enable the deconz.service (headless) or deconz-gui.service (gui)
> 
> https://github.com/dresden-elektronik/deconz-rest-plugin#headless-support

Link update: 
https://github.com/dresden-elektronik/deconz-rest-plugin#headless-support-for-linux