dresden-elektronik/deconz-rest-plugin #1270: No lights in headless mode

joubbi, 2019-02-15:
I'm running deconz on debian.
The server is headless and I run deconz as a systemd service according to the readme.

`root@assistant:~systemctl status deconz
_ deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: e
   Active: active (running) since Fri 2019-02-15 17:03:54 CET; 24min ago
 Main PID: 29694 (deCONZ)
   CGroup: /system.slice/deconz.service
           __29694 /usr/bin/deCONZ -platform minimal --http-port=80

Feb 15 17:03:54 assistant systemd[1]: Started deCONZ: ZigBee gateway -- REST API
Feb 15 17:03:54 assistant deCONZ[29694]: libpng warning: iCCP: known incorrect s
Feb 15 17:03:54 assistant deCONZ[29694]: This plugin does not support propagateS
Feb 15 17:03:54 assistant deCONZ[29694]: This plugin does not support propagateS
`

The webpage is accessible and works. I can reach it on port 80.
There are no lights or other devices there however.
There is nothing listening on port 8080.

If I run `root@assistant:~systemctl stop deconz
root@assistant:~# /usr/bin/deCONZ -platform minimal --http-port=80
libpng warning: iCCP: known incorrect sRGB profile
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
`
Then the page is accessible on both port 80 and 8080 with lights and everything.
If I don't stop the service before running deCONZ, port 80 won't have the lights but 8080 will.

What is going on?

manup, 2019-02-15:
Looks like you have two instances of deCONZ running?

    ps ax | grep deCONZ

joubbi, 2019-02-15:
```
root@assistant:~# systemctl stop deconz
root@assistant:~# systemctl start deconz
root@assistant:~# ps ax | grep deCONZ
  249 ?        Ss     0:30 /bin/bash /usr/bin/deCONZ-update2.sh
 7528 ?        Ssl    0:00 /usr/bin/deCONZ -platform minimal --http-port=80
 7599 ttyS0    S+     0:00 grep deCONZ
```

manup, 2019-02-15:
Yes but how is the output before stopping the service?

deCONZ can only listen on one port per instance, so if it is available on port 80 and 8080 it seems that two instances are running.

joubbi, 2019-02-15:
I stopped the service. Then:
```
root@assistant:~# ps ax | grep deCONZ
  249 ?        Ss     0:30 /bin/bash /usr/bin/deCONZ-update2.sh
 8698 ttyS0    S+     0:00 grep deCONZ
root@assistant:~# systemctl start deconz
root@assistant:~# ps ax | grep deCONZ
  249 ?        Ss     0:30 /bin/bash /usr/bin/deCONZ-update2.sh
 8706 ?        Ssl    0:00 /usr/bin/deCONZ -platform minimal --http-port=80
 8739 ttyS0    S+     0:00 grep deCONZ
root@assistant:~# 
```
There are no lights.
It's only listening on port 80.

manup, 2019-02-15:
I mean how is the output of `ps ax | grep deCONZ` **before** stopping the service?

joubbi, 2019-02-15:
I don't understand what you mean.
```root@assistant:~# # start from zero
root@assistant:~# ps ax | grep deCONZ
  249 ?        Ss     0:30 /bin/bash /usr/bin/deCONZ-update2.sh
10936 ttyS0    S+     0:00 grep deCONZ
root@assistant:~# systemctl start deconz
root@assistant:~# ps ax | grep deCONZ
  249 ?        Ss     0:30 /bin/bash /usr/bin/deCONZ-update2.sh
10944 ?        Ssl    0:00 /usr/bin/deCONZ -platform minimal --http-port=80
10976 ttyS0    S+     0:00 grep deCONZ
root@assistant:~# tmux
```
In tmux:
```
root@assistant:~# /usr/bin/deCONZ -platform minimal --http-port=80
libpng warning: iCCP: known incorrect sRGB profile
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
```
Then jumping back out from tmux.
```
root@assistant:~# ps ax | grep deCONZ
  249 ?        Ss     0:30 /bin/bash /usr/bin/deCONZ-update2.sh
10944 ?        Ssl    0:09 /usr/bin/deCONZ -platform minimal --http-port=80
11581 pts/0    Sl+    0:01 /usr/bin/deCONZ -platform minimal --http-port=80
11765 ttyS0    S+     0:00 grep deCONZ
root@assistant:~# 
```

manup, 2019-02-15:
>I don't understand what you mean.

This:

    10944 ?        Ssl    0:09 /usr/bin/deCONZ -platform minimal --http-port=80
    11581 pts/0    Sl+    0:01 /usr/bin/deCONZ -platform minimal --http-port=80

You have two instances of deCONZ running.

I just noticed when you're starting deCONZ as root, which is different from what the systemd service does. The default deconz.service file will start deCONZ as user with the id 1000.

So if the lights are available when you are running as root it means all the files including zll.db are likely under /root/.local/share ...

So you need to either:
* Copy these files in the related home dir of the user with id 1000 (also correct ownership)
* Or adapt the service file to use user root instead 1000 (which I wouldn't recommend)

joubbi, 2019-02-15:
Ahh.
I didn't know that the service is running as UID 1000.
Once you mentioned it I realized that the user probably don't have permissions to read the USB device.
I did this to fix the problem:
```
usermod -a -G dialout username
```
I think it would be helpful if this is mentioned in the README.md.

Thank you for the help!

closed 2019-06-22 by stale[bot]