dresden-elektronik/deconz-rest-plugin #1271: Issue with SYLVANIA SMART+ BR30 Tunable Bulbs becoming unreachable.

Elliott32224, 2019-02-15:
I have six SYLVANIA SMART+ BR30 Tunable White LED Bulb. Up until deConz Windows version 2.05.58, they were working quite well and stayed connected. With version 2.05.58, 4 of the 6 randomly became unreachable and at least one spontaneously lit up. I was hoping that 2.05.59 would have fixed this problem. But it has not. I wonder if anyone else is having this issue.

manup, 2019-02-15:
It might be worth to use the same attribute reporting configuration for OSRAM/SYLVANIA as we do for IKEA lights, in order to keep the traffic and tasks for the lights on low profile.

Can you please provide some screenshots of one of the bulbs as described in https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

Mainly Node Info and Basic Cluster with model identifier and manufacturer name are interesting.

Elliott32224, 2019-02-16:
Thanks for the quick response. I have attached a file with the information that you requested. There are data for 2 different light bulbs, both OSRAM. Each has 2 views, because of scrolling issues for the cluster window. The first light (light 2) was reachable this morning.The second light (light 3) was not. I don't see any differences between the two with respect to the information. Thanks for your help.
[deConz issue.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2871859/deConz.issue.pdf)

Elliott32224, 2019-02-19:
Thanks for your help. The problem was solved when I manually updated the firmware of my Conbee by the procedure you posted elsewhere on GitHub. It seems that it is not good to have Phoscon software and Conbee firmware out of phase with respect to updates. By the way, I had to do the firmware update on a different computer from the one running deConz, because the software could not find the correct USB port for the Conbee (too many devices in different ports). Elliott

Elliott32224, 2019-02-20:
Spoke too soon. This morning two of my OSRAM bulbs were unreachable. How do I do what you suggested earlier: _

> It might be worth to use the same attribute reporting configuration for OSRAM/SYLVANIA as we do for IKEA lights, in order to keep the traffic and tasks for the lights on low profile._

_

manup, 2019-02-22:
This needs to be changed in code, I'll see to bring these changes into 2.05.60 for OSRAm/SYLVANIA lights. I just can test with the OSRAM E27 and GU10 lights of the European version. In the coming weeks we get more US devices for testing.

Elliott32224, 2019-02-22:
Thank you!

________________________

Elliott Richelson

Jacksonville, FL 32224

*Confidentiality Notice: *This e-mail message, including any attachments,
is for the sole use of the intended recipient(s) and may contain
confidential and privileged information. Any unauthorized review, use,
disclosure or distribution, is prohibited. If you are not the intended
recipient, please contact the sender by personal contact or reply e-mail
and destroy all copies of the original message.

Elliott32224, 2019-02-26:
> This needs to be changed in code, I'll see to bring these changes into 2.05.60 for OSRAm/SYLVANIA lights. I just can test with the OSRAM E27 and GU10 lights of the European version. In the coming weeks we get more US devices for testing.

I wanted to mention that my OSRAM Lightify bulbs were working perfectly fine until around 2.05.57. This may give you a clue as to what needs to be fixed. Thank you.
Elliott

Elliott32224, 2019-03-12:
Upgraded software (V2.05.60) and firmware (26320500) yesterday and, unfortunately, OSRAM Lightify bulb problem still persists. Elliott

Elliott32224, 2019-04-06:
> This needs to be changed in code, I'll see to bring these changes into 2.05.60 for OSRAm/SYLVANIA lights. I just can test with the OSRAM E27 and GU10 lights of the European version. In the coming weeks we get more US devices for testing.

Anything new to report? Thank you.

closed 2019-08-11 by stale[bot]