dresden-elektronik/deconz-rest-plugin #1272: [Device Support Request] Sinope 600W Smart Dimmer - DM2500ZB

ericvandamme, 2019-02-16:
Hello,

I am requesting support for the [Sinope 600W Smart Dimmer](https://www.sinopetech.com/en/product/smart-dimmer-600-w-zigbee/). I have managed to make everything work via deCONZ, but it won't pair through Phoscon. It is, as far as all my searching has came up with, the only reasonably priced Zigbee switch certified for Canada Electrical code using 120V. It is also a very strong repeater with a strong connection to all Zigbee devices in my home and its compact form factor easily fits in the smaller electrical boxes found in older homes.

*Special Note, it wouldn't join the network unless I read Basic info during the pairing process. I would be more than happy to repeat the process to provide any additional information as required.

Thank you very much for your consideration.
[image]
[image]
[image]

manup, 2019-02-16:
Some Zigbee details in the SmartThings device driver: 

https://www.sinopetech.com/wp-content/uploads/2018/05/Sinope-Technologies-DM2500ZB-SVN-365.txt

To my understanding:
* The device can be supported as dimmable light
* Supports reporting for On/Off and Level Control cluster 
* Does not provide capabilities to remote control other lights

ericvandamme, 2019-03-16:
Tested with DeCONZ in version 2.05.60 and the device is working as expected.

closed 2019-03-16 by ericvandamme