dresden-elektronik/deconz-rest-plugin #1273: Can't connect Hue Apps

capstan1, 2019-02-16:
Hi,

when researching about Deconz there is always mentioned that you can use different Hue apps (I've read about the official one, all4Hue and HueEssentials) with the raspbee. However, only HueEssentials is working for me.

The official Hue App won't find the gateway, even if I enter the IP manually.
All4Hue is finding the gatway, but I can't establish a connection. I got in contact with the developer, who is asking what hue bridge firmware the phoscon gateway reports. I can't find that info anywhere. Can you give me a hint?

Also, any idea for the official app?

ebaauw, 2019-02-16:
> The official Hue App won't find the gateway, even if I enter the IP manually.

Are you running deCONZ on port 80?  The Hue app won't accept another port.  For the Hue app to find the gateway, make sure UPnP is not disabled when starting deCONZ.

manup, 2019-02-16:
Did you also enable *Authentificate app* in the gateway settings, otherwise all4Hue won't be able to connect. When I remember correctly it also only support deCONZ running on port 80.

https://doc.phoscon.de/app/doc.html#settings-gateway-advanced-en

Haven't tried the official Hue app in a long time, not sure if it still works anymore.

capstan1, 2019-02-16:
I did click on Authetificate app in the settings. For the port part: actually not sure. Im running [this Hass.io Add-on](https://github.com/marthoc/hassio-addons/tree/master/deconz) with standard settings.

KodeCR, 2019-02-18:
The official Hue App needs ssl with a certificate and bridge-id based on the lan mac address. See https://github.com/KodeCR/hue-pass. Not sure if it's possible to run that with Hass.io though.

pponce, 2019-03-22:
@KodeCR 
Thank you so much. I was pulling my hair out trying to figure out why i could get the hue app to discover my deconz gateway but not actually authenticate and pair.  hue-pass did the trick!

KodeCR, 2019-05-10:
You're welcome. It would be nice if deconz supported https with the proper certificate natively at some point, but I've been using hue-pass for months now without issue :)

pponce, 2019-05-13:
Today i've been trying to connect to the phoscon gateway via the philips hue app but i'm not able to.
Not sure if hue-pass has stopped working with deconz or if it's something else.

KodeCR, 2019-05-15:
I found that sometimes after a reboot the hue-pass service starts to soon. If I stop the hue-pass service and start it again manually, that normally fixes. I'll see if I can delay the service start a bit.

pponce, 2019-05-15:
ah ok.  Will give that a try.  Does deconz have a .service start up file?  might be able to add something like "after=deconz.service" in the hue-pass.service file.

closed 2019-09-19 by stale[bot]

coke7, 2020-10-25:
Hi,
I had the same problem - no bridge, no 172.30.33.0.
After I set the api port 40850 - still no bridge...
Then I tried to change my gateway name and voila the bridge was found automatically. My gateway had a minus "-" before it worked, so I think there are some characters causing problems within the hassio deconz combination.
Hope this helps some of you

theCroh, 2021-01-10:
> Hi,
> I had the same problem - no bridge, no 172.30.33.0.
> After I set the api port 40850 - still no bridge...
> Then I tried to change my gateway name and voila the bridge was found automatically. My [...]

What kind of Home Assistant installation do you have? HASS.IO or something else?

coke7, 2021-01-10:
> What kind of Home Assistant installation do you have? HASS.IO or something else?

Exactly - the image for the pi model 3b 32 bit

answer35, 2021-02-11:
Hi, I currently run Hass.io on docker using the Deconz addon. I am not sure to understand how to setup everything right so I can connect with Hue app on my Conbee gateway. Is anyone can help me configuring this please ?

7wells, 2021-03-02:
Hello!
So the official Philips Hue app cannot connect out-of-the-box to a RaspBee II running deconz and phoscon?

Smanar, 2021-03-02:
Good question ^^.
I know some time ago it was possible, but idk ATM. I have see some change on the code recently on that, but I have never used smartphone app on it https://github.com/dresden-elektronik/deconz-rest-plugin/commit/19a0e7032348321a74ff238bc2d5ba448df73a01

7wells, 2021-03-04:
@YKO-de or any of the developers:
Do you have an idee if/how the official Philips Hue app can connect out-of-the-box to a RaspBee II running deconz and phoscon? Thanks for sharing your knowledge. 👍

manup, 2021-03-04:
Historically the deCONZ REST-API shares some similarities with the Hue API and "sometimes" the Hue app works _partially_ with it, however this is a cat and mouse game and not actively / officially supported — the Hue App might work on one day and breaks with the next update.

I would rather recommend the Hue Essentials app which has official deCONZ REST-API support.