dresden-elektronik/deconz-rest-plugin #1275: Can't change Zigbee channel

PalmMx, 2019-02-17:
I'm trying to change the Zigbee channel on my network, but my attempts are without success.

I’m running a ConBee USB stick on a Raspberry Pi 3, with Hassio. I’m using the addon by Marthoc. deCONZ version is 2.05.59 and ConBee firmware is 262F0500.

It seems like the network is stuck to Channel 11, but I want to use Channel 25 instead. However, resetting the gateway and/or trying to do a channel change from the Advance Settings menu won’t do. I have tried both the Phoscon and the Old WebApp version (2016) UI. Nothing happens and when I refresh it goes back to Channel 11 all the time.

What am I missing here? What shall I do to change the Zigbee channel?

PalmMx, 2019-02-24:
Didn't have the time to wait for a solution on the channel issue, so I unplugged the ConBee from the Raspberry Pi and connected it to my PC (Windows 8.1) and downloaded the deCONZ software. I was (of course) able to changed the Zigbee channel to Channel 25 on my PC. Then I disconnected the ConBee from the PC and re-connected it on the RPi. Channel 25 was kept. Case closed.

closed 2019-02-24 by PalmMx

MickL, 2019-06-18:
I have the same problem with my RaspBee. The channel switching shows an endless loading animation. I am stuck on channel 0 and cant find a device. Also i cant reset the Gateway. Could you please reopen?

MickL, 2019-06-18:
Nevermind I had to edit the config.txt in the root of the sd card: 
https://github.com/marthoc/hassio-addons/blob/master/deconz/RASPBEE-SETUP.md

This file cant be edited from HASSIO configurator, it has to be done on the sd card. Also macOS cant read the filesystem/sd-card.