dresden-elektronik/deconz-rest-plugin #1276: Add daylight control

ronaldvans, 2019-02-18:
While using Phoscon I can't add a daylight control using the build in sensor. In "Create sensor control" and selecting "Daylight control" the Create button is grayed out irrespective of text in the Name box.

choogendoorn, 2019-03-08:
Yup, same here

sajox, 2019-04-07:
Same here. 
Also tried to enable the "network service" in the old webapp without result. Reboot doesent help either.

RDG88, 2019-04-08:
Same for me

ronaldvans, 2019-04-09:
@manup, any ideas how we can enable this through the Phoscon app?

manup, 2019-04-09:
Hi, sorry the button is misleading and will be removed or hidden in next versions.

The truth is this feature isn't implemented in the Phoscon App yet. There is only an older version of a daylight control in the old WebApp but I don't recommend it anymore.

Albeit a new proper daylight control is on the todo list for a long time now, we don't have an ETA for it.

closed 2019-04-10 by ronaldvans

pilotniq, 2019-11-19:
The button is still there in the latest version which I installed today, November 19, 2019.