dresden-elektronik/deconz-rest-plugin #1277: Show groups of class room in Phoscon

KodeCR, 2019-02-18:
Could you please add support to show groups of class "Room" in Phoscon and not just LightGroups. Thanks.

KodeCR, 2019-06-19:
All that is needed to support this is to:
- change '"LightGroup"===group.get("type")' to '("LightGroup"===group.get("type")||"Room"===group.get("type"))' in js/devices_lights.js, js/main.js, and js/switch-wizard-editor.js
- change '"LightGroup"!==x.get("type")' to '"LightGroup"!==x.get("type")||"Room"!==x.get("type")' in js/devices_switches.js
- change '"LightGroup"!==group.get("type")' to '"LightGroup"!==group.get("type")&&"Room"!==group.get("type")' in js/group.js

Thanks

manup, 2019-06-19:
Thanks I'll put it on the list

closed 2019-10-24 by stale[bot]

junemire, 2021-01-15:
It think this was never built-in, correct? If so, I was wondering if that is still something relevant on the list? From my experience there is a lot of vendors and clients that work on a room basis. But these views and lists are always empty (in client apps) as I setup all my settings/scenes on light groups in Phoscon. Like the idea to have everything managed/showing up in the bridge UI vs only in "client apps".. Phoscon web app has a great UX in opinion...

aschmidt74, 2022-07-06:
Just realized that `rooms` created in the `Hue Essentials` app dont show up in Phoscon. `groups` are visible.
I'd like to see both and group them accordingly! **feature-request** ;)

Mimiix, 2022-07-06:
Phoscon feature requests are done on the forums :)