dresden-elektronik/deconz-rest-plugin #1278: Zigbe network fails and asks for firmware update every day

cobirnm, 2019-02-18:
I have a conbee setup with docker. Every few hours my zigbee network stops. When I connect using VNC I get an update firmware needed. If I push the update firmware button it updates and then the network is up and running for some hours. After some hours again the network stops and I get again the update firmware message using VNC to conect.
If I go to Phoscon app I see that firmware is up to date. Can you please help me.

manup, 2019-02-18:
Doy you have any other usb devices connected to the machine?

Which deCONZ and firmware version is shown in the Phoscon App?

cobirnm, 2019-02-18:
Hello,

When this happens I get in phoscon that firmware is not conected. I have another usb device conected a zwave stick.

[image]

[image]

I'm updating manually via docker but I've done this for the past 3 days.

jcf6288, 2019-02-19:
I had a similar issue, make sure your CONBEE is set as the coordinator and not a router.

manup, 2019-02-19:
>I have another usb device conected a zwave stick.

If possible can you check if the problem stays when unplugging the Z-Wave stick?

cobirnm, 2019-02-19:
Hello,

I've had the z-wave stick unpluged for the day and again it has happen. It's always more or less every 24 hour. I don't now how to better describe this issue.
Today I've conected to deconz using vnc and started noticing that the app was disconecting. It stayed conected for 30 seconds and would disconect. This happened for about 15 minutes. After that it's stoped and asking for firmware update. My zigbee netword is still up but unresponsive.
This is a new conbee I've had it for a week. Can you please advise if I should return it.

manup, 2019-02-19:
It's very likely not the hardware but but something is hitting the usb connection. Is the service/Docker container which is accessing the Z-Wave stick also stopped?

Since this is a Docker installation can you check the following output if the usb devices are still listed (on the host):

    ls /dev/ttyUSB*

The stick should be still listed with the same number as configured in Docker like /dev/ttyUSB0

And

    lsusb

cobirnm, 2019-02-19:
At this point and after some tests I'm able to see that there is some kind of incompatability with the z-wave stick. As soon as I plug it in the deconz under VNC (sorry I still miss the correct app words) starts disconecting and reconecting. 

I've checked the config and Zwave stick is at /dev/ttyACM0 and Conbee at /dev/ttyUSB0. At this point I'm going crazy.  Do you know of similar issues?

cobirnm, 2019-02-20:
These are my logs:

12:31:54:573 device state timeout   ignored in state 3
--
12:31:55:524 device state timeout   ignored in state 3
12:31:56:473 device state timeout   ignored in state 3
12:31:57:431 device state timeout   ignored in state 3
12:31:58:430 device state timeout   ignored in state 3
12:31:59:430 device state timeout   (handled)
12:31:59:714 0x00158D0001B765A3   error APSDE-DATA.confirm: 0xA7 on task
12:31:59:929 void   zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
12:31:59:929 device state timeout   ignored in state 1
12:32:00:494 dev /dev/ttyAMA0

at this point it still holds. I need to wait for about 24 hours before it asks again for firmware updarte.

Philje123, 2019-02-20:
Have you tried updating the firmware manually in windows using GCFFlasher?

It could be that the firmware update isn't actually taking.

Use the latest firmware version from here (took me a while to find this page):

https://www.dresden-elektronik.de/rpi/deconz-firmware/?C=M;O=D

I had a problem where the firmware update wasn't working at it kep saying the same as yours (Not connected). Updated it manually using GCFFlasher and all is well.

cobirnm, 2019-02-20:
> > I've checked the config and Zwave stick is at /dev/ttyACM0 and Conbee at /dev/ttyUSB0. At this point I'm going crazy. Do you know of similar issues?
> 
> deCONZ shouldn't disconnect just because another dongle is plugged in, actually I do this quite often.
> 
> Can you please post the output of the two commands mentioned above when the disconnect happens?

the output is as follows:

pi@raspberrypi:~ $ ls /dev/ttyUSB*
/dev/ttyUSB0

pi@raspberrypi:~ $ lsusb
Bus 001 Device 004: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 005: ID 0658:0200 Sigma Designs, Inc.
Bus 001 Device 006: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

cobirnm, 2019-02-20:
I'm also including the logs from container:

17:40:34:504 remove dead link for   (5F68, 31DD)
--
17:40:35:580 remove dead link for   (0, 5F68)
17:40:39:880 remove dead link for   (5F68, CD58)
17:41:06:465 dev /dev/ttyAMA0
17:43:16:464 dev /dev/ttyAMA0
17:45:26:464 dev /dev/ttyAMA0
17:45:31:754 Announced to internet
17:45:31:754 discovery server date:   Wed, 20 Feb 2019 17:45:31 GMT
17:45:31:754     local time seems to be ok
17:47:36:464 dev /dev/ttyAMA0
17:49:46:464 dev /dev/ttyAMA0
17:51:56:507 dev /dev/ttyAMA0
17:54:06:508 dev /dev/ttyAMA0
17:55:31:712 Announced to internet
17:55:31:713 discovery server date:   Wed, 20 Feb 2019 17:55:31 GMT
17:55:31:713     local time seems to be ok
17:56:16:468 dev /dev/ttyAMA0
17:58:26:508 dev /dev/ttyAMA0
18:00:36:470 dev /dev/ttyAMA0
18:02:46:517 dev /dev/ttyAMA0
18:04:56:508 dev /dev/ttyAMA0
18:05:31:687 Announced to internet
18:05:31:688 discovery server date:   Wed, 20 Feb 2019 18:05:31 GMT
18:05:31:688     local time seems to be ok
18:07:06:464 dev /dev/ttyAMA0
18:09:16:503 dev /dev/ttyAMA0
18:11:26:498 dev /dev/ttyAMA0
18:13:36:508 dev /dev/ttyAMA0
18:15:31:695 Announced to internet
18:15:31:695 discovery server date:   Wed, 20 Feb 2019 18:15:31 GMT
18:15:31:695     local time seems to be ok
18:15:46:508 dev /dev/ttyAMA0
18:17:56:466 dev /dev/ttyAMA0
18:20:06:508 dev /dev/ttyAMA0
18:22:16:509 dev /dev/ttyAMA0
18:24:26:507 dev /dev/ttyAMA0
18:25:31:793 Announced to internet
18:25:31:793 discovery server date:   Wed, 20 Feb 2019 18:25:31 GMT
18:25:31:793     local time seems to be ok
18:26:36:507 dev /dev/ttyAMA0
18:28:46:516 dev /dev/ttyAMA0
18:30:56:509 dev /dev/ttyAMA0
18:33:06:516 dev /dev/ttyAMA0
18:35:16:509 dev /dev/ttyAMA0
18:35:31:738 Announced to internet
18:35:31:738 discovery server date:   Wed, 20 Feb 2019 18:35:31 GMT
18:35:31:738     local time seems to be ok
18:37:26:508 dev /dev/ttyAMA0
18:39:36:513 dev /dev/ttyAMA0
18:41:46:510 dev /dev/ttyAMA0
18:43:56:509 dev /dev/ttyAMA0
18:45:31:689 Announced to internet
18:45:31:689 discovery server date:   Wed, 20 Feb 2019 18:45:31 GMT
18:45:31:689     local time seems to be ok
18:46:06:507 dev /dev/ttyAMA0
18:48:16:509 dev /dev/ttyAMA0
18:50:26:463 dev /dev/ttyAMA0
18:52:36:465 dev /dev/ttyAMA0
18:54:46:464 dev /dev/ttyAMA0
18:55:31:753 Announced to internet
18:55:31:754 discovery server date:   Wed, 20 Feb 2019 18:55:31 GMT
18:55:31:754     local time seems to be ok
18:56:56:506 dev /dev/ttyAMA0
18:59:06:507 dev /dev/ttyAMA0
19:01:16:466 dev /dev/ttyAMA0
19:03:26:515 dev /dev/ttyAMA0
19:05:31:698 Announced to internet
19:05:31:699 discovery server date:   Wed, 20 Feb 2019 19:05:31 GMT
19:05:31:699     local time seems to be ok
19:05:36:515 dev /dev/ttyAMA0
19:07:46:465 dev /dev/ttyAMA0
19:09:56:497 dev /dev/ttyAMA0
19:12:06:511 dev /dev/ttyAMA0
19:14:16:511 dev /dev/ttyAMA0
19:15:31:685 Announced to internet
19:15:31:685 discovery server date:   Wed, 20 Feb 2019 19:15:31 GMT
19:15:31:685     local time seems to be ok
19:16:26:463 dev /dev/ttyAMA0
19:18:36:496 dev /dev/ttyAMA0
19:20:46:510 dev /dev/ttyAMA0
19:22:56:515 dev /dev/ttyAMA0
19:25:06:525 dev /dev/ttyAMA0
19:25:31:699 Announced to internet
19:25:31:700 discovery server date:   Wed, 20 Feb 2019 19:25:31 GMT
19:25:31:700     local time seems to be ok
19:27:16:467 dev /dev/ttyAMA0
19:29:26:497 dev /dev/ttyAMA0
19:31:36:463 dev /dev/ttyAMA0
19:33:46:516 dev /dev/ttyAMA0
19:35:31:731 Announced to internet
19:35:31:732 discovery server date:   Wed, 20 Feb 2019 19:35:31 GMT
19:35:31:732     local time seems to be ok
19:35:56:505 dev /dev/ttyAMA0
19:38:06:500 dev /dev/ttyAMA0
19:40:16:464 dev /dev/ttyAMA0
19:42:26:463 dev /dev/ttyAMA0
19:44:36:463 dev /dev/ttyAMA0
19:45:31:711 Announced to internet
19:45:31:712 discovery server date:   Wed, 20 Feb 2019 19:45:31 GMT
19:45:31:712     local time seems to be ok

By what I can see it's not trying to conect to usb

manup, 2019-02-20:
Interesting.

>By what I can see it's not trying to conect to usb

It seems that the /dev/ttyUSB0 is no longer available in the container but the host.

What happens if you just restart the container?

cobirnm, 2019-02-22:
Restarting the software solves for an hour or so but then the situation is the same.  I will try an installation that does not rely on docker but at this point I'm considering returning the product.

cobirnm, 2019-02-25:
I think I can now say that Conbee is incompatible wit systems using Aeotec zwave stick. In a raspberry pi using just Conbee everything is working as it should. I think that this issue can be closed.

Philje123, 2019-02-25:
I have an Everspring SA413 (basically a Sigma z-wave stick) and it's working perfectly with the conbee.

Did you try as I said and update the firmware manually (outside of docker etc)?

I'm running it on a raspberry pi running HassOS and it's working perfectly.

What are you running it on?

cobirnm, 2019-02-25:
I tried multiple firmwares including the last one available, 26300500. I ended using a spare raspberry py just for deconz.

closed 2019-07-02 by stale[bot]