dresden-elektronik/deconz-rest-plugin #1279: Phoscon behind ngnix: Login accepts any password

Andre0512, 2019-02-18:
Hello,

i tried to make my Phoscon WebApp reachable from outside with forwarding the port with a nginx-webserver. 

[image]
When I click on this Page to  _Raspbee_, I coming in 90% to the  _Set login password_-Page

When I insert a password, I always get _Login failed_, but I see in the js-console an api-link which I can use for the rest-api.
[image]

In 10% of cases (I assume this happens when I click multiple times on Raspbee, but it also could just be random), I came to the login page. I can enter any password, the login is **always** successfull.
[image]

Here is my raspbee config.
[image]

Here is my nginx configuration.
```
server {
        listen 443 ssl;
        listen [::]:443 ssl;
        server_name example.com;

        ssl_certificate /etc/letsencrypt/live/example.com/fullchain.pem;
        ssl_certificate_key /etc/letsencrypt/live/example.com/privkey.pem;

        location / {
                proxy_pass http://127.0.0.1:8080;
        }
}

```

What am I doing wrong?
Tell me if you need additional informations.

snozzlebert, 2019-02-19:
I think you need to add

```
        proxy_set_header Accept $http_accept;
        proxy_set_header Authorization $http_authorization;
```
after your proxy_pass line.

Andre0512, 2019-02-19:
Unfortunately that does not work. Still behaves as described.

corneyl, 2019-03-05:
For me it seems to work, I use the following config:
```
server {
	listen 443 ssl;

	server_name deconz.*;

	include /config/nginx/ssl.conf;

	client_max_body_size 0;
	proxy_buffering off;

	location / {
		include /config/nginx/proxy.conf;
                proxy_set_header Upgrade $http_upgrade;
                proxy_set_header Connection "upgrade";
		proxy_pass http://deconz:8080;
	}
}
```
Where `/config/nginx/proxy.conf` contains:
```
## Version 2018/05/31 - Changelog: https://github.com/linuxserver/docker-letsencrypt-armhf/commits/master/root/defaults/proxy.conf

client_max_body_size 10m;
client_body_buffer_size 128k;

#Timeout if the real server is dead
proxy_next_upstream error timeout invalid_header http_500 http_502 http_503;

# Advanced Proxy Config
send_timeout 5m;
proxy_read_timeout 240;
proxy_send_timeout 240;
proxy_connect_timeout 240;

# Basic Proxy Config
proxy_set_header Host $host:$server_port;
proxy_set_header X-Real-IP $remote_addr;
proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
proxy_set_header X-Forwarded-Proto https;
proxy_set_header X-Forwarded-Host $host;
proxy_set_header X-Forwarded-Ssl on;
proxy_redirect  http://  $scheme://;
proxy_http_version 1.1;
proxy_set_header Connection "";
proxy_cache_bypass $cookie_session;
proxy_no_cache $cookie_session;
proxy_buffers 32 4k;
```
Maybe it's one of the headers that fixes it.

closed 2019-07-10 by stale[bot]

soumya92, 2019-07-12:
If you're running nginx on the same device as deCONZ, make sure you use the *external* IP in the proxy block. That is,

```nginx
location / {
    proxy_pass http://192.168.1.182:8080
}
```

instead of

```nginx
location / {
    proxy_pass http://127.0.0.1:8080
}
```

It seems there is special handling for localhost that confuses the PWA when running behind the reverse proxy. The server thinks the request is trusted because it is coming from localhost, but parts of the JS assume a login is necessary because the URL in the browser is not localhost. This confusion leads to the unexpected behaviour described here.

flechaig, 2019-09-04:
This is how I managed to deal with this issue:
```
  location ~ (/.*) {
    set $p 80;
    if ($http_upgrade = "websocket") {
     set $p 443;
    }
    if ($http_upgrade = "") {
     set $p 80;
    }
    proxy_pass http://[censored IP address]:$p$1;
    proxy_set_header Host $host;
    proxy_set_header X-Forwarded-For $remote_addr;
    proxy_http_version 1.1;
    proxy_set_header Upgrade $http_upgrade;
    proxy_set_header Connection "upgrade";
  }
```
With this setup, Phoscon app works fine and websockets also. Be careful: if statement in NGINX costs some performance, so it is not suitable for high load instances.

morfei1, 2020-08-12:
> server {
> 	listen 443 ssl;
> 
> 	server_name phoscon.*;
> 
> 	include /config/nginx/ssl.conf;
> 
> 	client_max_body_size 0;
> 	proxy_buffering off;
> 
> 	location / {
> 		include /config/nginx/proxy.conf;
>                 proxy_set_header Upgrade $http_upgrade;
>                 proxy_set_header Connection "upgrade";
> 		proxy_pass http://192.168.0.x:80;
> 	}
> }

It was my way to run the reverse proxy with nginx on my setup, thanks to the @corneyl sharing.
Regards!

lweberru, 2021-09-14:
I am using the same config as @flechaig , (without ssl). And if I start a discovery it seems that it finds the device:
[image]
but there is still only the phosco logo on the screen, after the scan has finished and no selection as it is when I enter the internal URL.

my config finally looks like this:
```
server {
   server_name deconz.klue.de
   listen ${NGINX_PORT};
   proxy_buffering off;
   client_max_body_size 0;

   location  / {
       set $p 80;
       if ($http_upgrade = "websocket") {
         set $p 443;
       }
       if ($http_upgrade = "") {
         set $p 80;
       }
       proxy_pass http://192.168.1.153:$p$1;
       proxy_set_header Host $host;
       proxy_set_header X-Forwarded-For $remote_addr;
       proxy_http_version 1.1;
       proxy_set_header Upgrade $http_upgrade;
       proxy_set_header Connection "upgrade";

   }
}

```
Any tips?

flechaig, 2021-09-17:
> Any tips?

You may need to adjust the Content-Security-Policy. Look into your browser's console to see if some content is blocked by CSP.
Then add the needed URLs in this directive:
`add_header Content-Security-Policy "default-src 'self';`

lweberru, 2022-01-15:
Still dont get it up via reverse proxy. Whenever I click on the discovered device, it tries to connect to the local/private ip (192.168.1.xxx) directly instead of going threw the external dns url:
[image]
https://192.168.1.153:8080/api/config?_=1642247779494
it should be something like this
https://deconz.mydomain.tld/api/config?_=1642247779494
how can I configure this?

deluxestyle, 2022-06-13:
> Still dont get it up via reverse proxy. Whenever I click on the discovered device, it tries to connect to the local/private ip (192.168.1.xxx) directly instead of going threw the external dns url: [image] https://192.168.1.153:8080/api/config?_=1642247779494 it should be something like this https://deconz.mydomain.tld/api/config?_=1642247779494 how can I configure this? P.S.: the reverse proxy is running on a different machine than the deconz, but in the same network

I do have the same issue. Did you found any solution?

lweberru, 2022-06-14:
No i did not found any solution

nlincke, 2023-03-26:
I guess this issue is still there? I would also like to let it point to a fqdn of my choice...