dresden-elektronik/deconz-rest-plugin #1280: How can i change the id on a switch?

bjarq, 2019-02-18:
My setup is Home Assistant on a Raspberry Pi with hassos. 
My Raspbee is on the same Raspberry. deCONZ add-on and Node-RED add-on.
I've added two ikea tradfri on/off switches and they share the same name.
[image]
In Node-RED the deconz_event payload.event.id is the same as well.
[image]
How can i change this?

Yes I'm newbee :-)

/bjarke

manup, 2019-02-18:
This is a Home Assistant specific id (perhaps auto generated based on the name?)
deCONZ only uses numeric id's like "1" and uniqueid like "00:11:22:33:44:55:66:77-01-0006".

So this question might be better answered in the Home Assistant community.

bjarq, 2019-02-18:
Thank you I'll try that:-)

lbschenkel, 2019-02-23:
@bjarq Although this is off-topic here, I can enlighten things for you because I'm also a HASS user. This is how it works:
- in the *first* time an entity is seen/imported in HASS, the HASS entity id will be generated based on the name of the device in Phoscon
- once it has been imported, the entity id in HASS will no longer change, even if you rename the device in Phoscon
- you can change the entity id in HASS via the entity registry (but restart HASS when you do, the `deconz` entities disappear when renamed but they come back with the new name after a restart)
- HASS (the `deconz` component) knows which entity is which device by using the `uniqueid` from the REST API

I hope that makes this clearer for you.

bjarq, 2019-02-24:
@lbschenkel - Thank you.

I found the file combining the information in HASS,  and how to correct the name in Phoscon before the device is discovered in HASS. All my zigbee devices is now connected to the Raspbee gateway :-)

What I find is on-topic here is that this IKEA on/off switch is the only device created with the same name in Phoscon. The IKEA remote control, the lights and the outlet switches get an increasing number automatically added to there names when discovered in Phoscon prior to importing them to HASS.

closed 2019-07-01 by stale[bot]