dresden-elektronik/deconz-rest-plugin #1281: ZSHARK capture to file

ebaauw, 2019-02-18:
I'm running ZSHARK on a dedicated Raspberry Pi, but WireShark on my macBook.  I'd like to run ZHARK continuously, even when I'm on the road with my macBook.  Would it be possible for ZSHARK to write the packets directly to a `.pcapng` file, rather than streaming them over the network to WireShark?  Maybe getting rid of the ZigBee encapsulation, to see the IEEE 802.15.4 Acks correctly?  Maybe with an option to start a new file every x packets or x hours?

manup, 2019-02-18:
Perhaps this can be done, with libpcap. There will be a small update to ZSHARK in the coming weeks, I'll put it on the check list.

Note you can also run Wireshark directly on the Raspberry Pi and just save the file there or better to  an usb storage :)

ebaauw, 2019-06-26:
Thanks.

Is zshark available thru the apt repository?

manup, 2019-06-26:
It's not yet included, but it will be in the near future for all supported platforms.

closed 2019-10-31 by stale[bot]