dresden-elektronik/deconz-rest-plugin #1282: WiFi password unencrypted in zll.db

ebaauw, 2019-02-18:
I noticed the `config2` table in `zll.db` contains my WiFi password unencrypted, even though I never let deCONZ manage my WiFi settings.  I configured the WiFi myself, setting up the Raspberry, before installing deCONZ for the first rime.  I actually leave it off most of the time, using the wired ethernet connection instead.

manup, 2019-02-18:
Strange this shouldn't happen, only configurations done via Phoscon App should be there (and these t will soon be encrypted with the underlying iw tools).

I'm not sure if this is an artefact of the in-between versions, need to check this with the colleagues.

SwoopX, 2019-02-20:
I can confirm that. Additionally, it seems that the gwpassword is stored as MD5, not the preferred way. You may go for SHA-512?

manup, 2019-02-20:
We're looking into this, format need to be supported by the wpa_supplicant and hostapd services.

ebaauw, 2019-06-21:
Just checked: in my most recent clean installation of deCONZ (for the ConBee II), the WiFi password is empty.  However it's still these in my older installations, now upgraded to v2.05.66.  The fix for this is taking too long for my taste; I cleared the field manually in the database.

manup, 2019-06-21:
To my knowledge the unencryped fields should be replaced by encryped versions when updating from older installations. I figure this is only done when WiFi is configured via the Phoscon App and we need to add code to clear existing passwords if that is not the case. I'll forward it to the maintainer.

closed 2019-10-26 by stale[bot]