dresden-elektronik/deconz-rest-plugin #1283: Cannot create a backup

iamkarlson, 2019-02-18:
Hi, I'm using version 58 and I haven't tried to create a backup in any other version. I go to the settings and click backup and then nothing happens but red cross with a "close" button. How can I find an issue and possibly solve it?

manup, 2019-02-19:
Can you also try with 2.05.59?

If it still doesn't work some more details would be helpful.

* Operating system
* If Linux which is the environment?
  Docker, Hass.io or started via systemd gui or headless mode?

iamkarlson, 2019-02-21:
Hi, 
After some investigation, I found that RaspBee was out of the network. So, that's the reason. However, it would be great if I know that from the error message rather than digging into all system components. What do you think?

manup, 2019-02-21:
Indeed, that should be improved. The reliance on connected Zigbee network isn't needed anymore since the related data is stored in the database anyway.

Will put this on the todo list.

iamkarlson, 2019-02-26:
Thanks @manup

closed 2019-02-26 by iamkarlson