dresden-elektronik/deconz-rest-plugin #1286: Support for Sunricher wall controls

OleEide, 2019-02-21:
Hi,
Is there any plans to add support for the Sunricher wall controls (https://www.sunricher.com/zigbee-series/zigbee-remotes.html).  I see the page lists some, but the one i have is named SR-ZG9001K4-RGBW, and SR-ZG2801K2-5C, both of the are RGBWCCT remotes,
I added the SR-ZG9001K4-RGBW using the Phoscon app, but never received any acknowledge, but the device showed up in the windows app, and there i was able to bind On/Off, Level Control and Color Control to an Philips RGBW bulb, and it works fine, but i need to be able to pick up the switch events from the winsock/rest api.  Is there any plans to add the Sunricher devices in the (near) future ?

/Ole

wvuyk, 2019-02-21:
Looking interesting. Follow the instructions on https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support to request support?

OleEide, 2019-02-21:
Reading trough the device request page, I hope I got it right, this is for the SR-ZG9001K4-RGBW:
[image]
[image]

/Ole

OleEide, 2019-02-21:
And this is for the SR-ZG2801K2-5C
[image] 
[image]

Both devices, is very interesting for us europeans, as the y fit perfectly with standard 55mm switch frames, the functions are similar, but with slightly different design.

/Ole

daaaaa, 2019-02-25:
Their products seem to be very interesting.

@OleEide Where did you buy them? I cannot find any store selling their items.

OleEide, 2019-02-25:
Yes they are quite interesting as they fulfill a demand for retrofitting existing switches.

@daaaaa 
I bought them in Norway at Tronika.no, don´t know if they ship outside Norway though..

But you may be able to order them straight from the manufacturer...

/Ole

ebaauw, 2019-02-25:
They look straightforward enough to support, but we need to know which command they send on which button press and whether they choose a group at random, or need to be bound to a group.  Running deCONZ with `--dbg-aps=2` should reveal that in the log.  Be sure to post the data indication as well as the asdu messages.

mathos77, 2019-04-14:
Related I suppose, still would love to see these work !
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/391

flemmingss, 2019-10-17:
> /Ole

How is the Sunricher wall controls working for you now? I am considering ZG9001T4-RGBW

zyp, 2019-11-07:
I picked up a SR-ZG9001K8-DIM tonight and like Ole I managed to get it joined to the network and bound to a light group, so I've got basic functionality in place, but naturally I would also like full support.

[image]
[image]
[image]

Note that mine are rebranded by the Norwegian importer, so the manufacturer name and the model identifier is different. If the model identifier is required to recognise the device, I can look up what they are for the other devices as well, so support can be added for all the rebranded ones along with the originals.

caplam, 2019-12-07:
Hello,
i'm a noob with zigbee. Starting with a raspbee, an osram smartplug and rgbw zigbee3 remote.
I bought this remote on iluminize.de but it's an oem of sunricher. 
In deconz it's identified as sunricher ZG2819S-RGBW.
I have no idea on what to do to learn more about and how to bind commands. When paired the remote is recognized as 4 Home automation color dimmer switch. 
My final goal is to catch commands sent by the remote and translate it (with jeedom) to a knx device (actually a dali device trough a knx/dali gateway).

closed 2020-04-12 by stale[bot]

jensflorian, 2020-06-11:
I'd like to know if the ZG9001T4 is supported?

https://www.sunricher.com/media/resources/manual/SR-ZG9001T4-RGBW-EU%20instruction%20V1.pdf

pur3asdf, 2020-09-02:
I'm also looking forward to see Sunricher remotes implemented into phoscon app. I can add my  [SR-ZG2868](https://www.sunricher.com/dim-cct-rgb-3-in-1-zigbee-remote-sr-zg2868.html) to the network and see it in deconz but i'm unable to add it as a switch in phoscon app.

Mimiix, 2020-09-03:
@jensflorian Please search in this git repo to check if theres a issue about it.

@pur3asdf Open a Device request in Phoscon Repo

rvanmoorsel, 2020-10-03:
Hi there,
I'm the owner of two types of Sunricher switches. One is the on/off and is detected in Phoscon as "Vendor: Sunricher / Product: Remote Controller". In HA it is: "ZGRC-KEY-007 by Sunricher". The other type has four rows of buttons. On/off - bright / dim... In Phoscon it's the same device and in HA it is : "ZGRC-KEY-009 by Sunricher". Both types pair to Deconz without problems and I paired them with the lights they need to switch.
The problem I have with it, is that they do not appear to send events from Deconz. I had an old setup in which this worked flawlessly. I could read deconz_event in HA and if I pressed the on button, an event was fired. It had the Switch ID in it and (when 'on' was pressed) event 1002. Unfortunately I don't know the versions anymore, because I tried ZHA as an alternative. But I still have the old config files, so I'm 100% sure this worked. This ZHA setup lasted just three weeks, because it was extremely unstable, opposed to the year of Deconz before that. 
When I google around for this issue, I can tell more Europeans are using these switches, because they fit nicely in the existing wall switches over here. I seems a lot of brands whitelabel it (mine says YPHIX as brand), but underwater it's always Sunricher.

By now I'm really disappointed I took the ZHA step, because the events aren't visible anymore. Is there anything that I can change to get the events back? Do you guys need more information from me? Or..?? I'm happy to help to supply this information, just let me know what you need.

Thanks a lot for all the hard work!

SwoopX, 2020-10-04:
Hey,

you should open 2 new issues (one per device),name them like "Missing button map for ZGRC-KEY-009", flag them as bug and provide the requested info. There's been quite some development around that lately. I assume what happend in your case as I recall, there was a rather generic button map assignment for sunricher devices. While tailoring that for known devices, that functionality got lost for others (we went from modelid starts with to modelid equals). On the other hand, we now provided users the means to modify assigned button maps to a device their selves and with the upcoming release, allow to define which button map to assign to which device. So it all goes into the direction becoming a kind of self service ;)

Pretty much al what's left for you to do is starting deconz with debug output enabled and start pushing all the buttons in all various combinations. Not the presses down and search the log for entries starting with "[INFO] - No button map for XYZ..." for your device. That information can then be easily translated into the required button map and later on included in the official release.

rvanmoorsel, 2020-10-08:
Hey SwoopX,
Thanks for the clarification. I'm not a hero with reporting bugs, but the way you describe it, I guess I can manage.
One thing, though. I'm currently not seeing anything happening when I use the 'listen to deconz_event' option in HA. That won't be a problem?
Thanks again, and I'll create these issues asap.