dresden-elektronik/deconz-rest-plugin #1287: Configuration reverts

titanous, 2019-02-22:
Whenever I restart deCONZ, the names of my devices/groups revert to a previous configuration (it's always the same). Not sure what debugging information would be helpful, there's no information in the logs that is obviously useful.

Version: 2.05.59

manup, 2019-02-22:
Which operation system/environment is deCONZ running in?

Might be related to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1288

titanous, 2019-02-23:
Linux amd64, using the non-HASS Docker image.

baranovoleg, 2019-03-01:
I am very excuse, but @titanous may you review this https://github.com/tent/http-link-go/pull/2 ?

rafadesu, 2019-05-18:
I have the same issue. Let me know if I can help debugging it.

titanous, 2019-07-20:
Root cause figured out in #1697.

closed 2019-07-20 by titanous