dresden-elektronik/deconz-rest-plugin #1288: deCONZ doesn't accept key after restart

Kane610, 2019-02-22:
Hi @manup 

Sometimes users get a really weird behaviour in deCONZ that after a restart of deCONZ it doesn't accept previously registered key. Reregistering with deCONZ after restart makes everything work again.

https://community.home-assistant.io/t/deconz-integration-hassio-0-88-all-disappeared-help/100691/12

skeptizzle, 2019-02-22:
I'm one of the folks having this issue in the link above.

manup, 2019-02-22:
I've checked the code, looks ok so far.
When a new api key is generated it is saved in the database in the next 5 seconds.

Can you check that the key is in the database after creation:

    sqlite3 zll.db 'select * from auth where apikey="B5EE783573"'

(replace with your apikey)

Then reboot, and check it it is still there.

skeptizzle, 2019-02-22:
@manup where is the database and where do I input the command?

Thanks for your time!

manup, 2019-02-23:
Just a quick check, I've logged in the hassio environment via ssh as described in https://developers.home-assistant.io/docs/en/hassio_debugging.html

To check where the zll.db is located:

    docker inspect -f "{{ .Mounts }}" addon_core_deconz
 
Accroding to the output the zll.db file is located in:

    /mnt/data/supervisor/addons/data/core_deconz/.local/share/dresden-elektronik/deCONZ

skeptizzle, 2019-02-23:
Okay so, sorry for the late reply..
If i inspect the container for mounts i get:

```
root@dockerhost:~# docker inspect -f "{{ .Mounts }}" addon_core_deconz
[{bind  /usr/share/hassio/addons/data/core_deconz /data  rw true rprivate} {bind  /lib/modules /lib/modules  ro false rprivate}]
```

I get this when looking up the api key from core.config_entries in the zll.db
```
root@core-deconz:/# sqlite3 zll.db 'select * from auth where apikey="D35620B975"'
Error: no such table: auth
```

What's next, @manup? Your help is much appreciated, i already learned some nifty commands!

Cheers!

hekm77, 2019-02-25:
I had same issue. HA 0.88.1. 
After removing the custom_updater component, deCONZ is working fine again.

skeptizzle, 2019-02-25:
@hekm77 great find, have you reported it over there? I will try this tomorrow and report back..

Cheers

hekm77, 2019-02-25:
@skeptizzle , This is not my find

https://community.home-assistant.io/t/deconz-official-thread/100504/6

skeptizzle, 2019-02-25:
@hekm77 ok, but thanks anyway. It was nice of you to reply here.

Kane610, 2019-02-28:
Some of these issues where affected by custom updater and the developer have taken steps to improve its performance. But there also was a bug with deCONZ component retry mechanism if getting a time out on start up. This will be fixed with Hass 0.89

Kane610, 2019-03-10:
No action for 10 days, closing this issue

closed 2019-03-10 by Kane610

nographix, 2020-11-15:
I have the same issue using homebridge
homebridge is running on RPi3
deconz is running on RPi4

does anyone know how to save the apikey?
The key gets lost after deconz restart.
Every power outage means that I cannot access to my zigbee network thru homebridge/homekit

I checked the database:
> sqlite3 zll.db 'select * from auth where apikey="xxxxxxxxxx"'
and got:
> Error: no such table: auth

Mimiix, 2020-11-16:
That means the DB isn't there/not complete

Anyway, please open your own issue, as this issue is way to old @nographix