dresden-elektronik/deconz-rest-plugin #1289: [Request device support] Sinope TH1300ZB floor heating thermostat

warrenmr, 2019-02-22:
https://www.sinopetech.com/en/product/smart-floor-heating-thermostat-3600-w-zigbee/

warrenmr, 2019-02-23:
I will post more details once I receive the deCONZ stick.

warrenmr, 2019-03-17:
[image]

This is where the temperature is set:

[image]

Node info:

[image]

justinfanok, 2019-06-17:
Would love this as well

closed 2019-10-22 by stale[bot]

the-speedmax, 2019-11-21:
Hello,
I already own a Conbee II. I'm looking to add some thermostat to my setup.
Do you know if this model can work with deconz? Sinope TH1123ZB & TH1124ZB 
These are Zigbee. 
https://www.sinopetech.com/en/product/smart-thermostat-for-electric-heating-3000-w-zigbee/?setLang=en
Tks for your help!