dresden-elektronik/deconz-rest-plugin #1290: All light entries disappeared from the gui (2.05.59)

PremiumUsername, 2019-02-22:
This is what happened:
I went into the zigbee configuration with Alt + Click on the Advanced Settings.
I tried to copy the network key, this activated the load function instead (is the whole area a button, or did I press the button on the right by mistake?). It loaded the latest version, the one created by the update to 2.05.59.
After it was done there were no more lights in the phoscon gui, even after a reboot.  Remotes, light groups (not functional since there are no lights) and sensors are still there.

The lights are still 'there' in the rest api and I can still use them in HA.

manup, 2019-02-22:
>I tried to copy the network key, this activated the load function instead (is the whole area a button, or did I press the button on the right by mistake?).

That's a bug, already fixed and will be part in the next version.

>After it was done there were no more lights in the phoscon gui, even after a reboot. Remotes, light groups (not functional since there are no lights) and sensors are still there.

Do you mean deCONZ GUI or the Phoscon App?

If on the Zigbee configuration page the parameters differ between rows (PANID, Channel or Network key) please revert back to the former configuration.

PremiumUsername, 2019-02-22:
Phoscon App

PremiumUsername, 2019-02-22:
...also the parameters does not differ. This is not a a huge deal for me, I use HA for control, so everything still works for me, but I can't use the phoscon app anymore.

manup, 2019-02-22:
Hmm, that smells like an Javascript Error.

When in the Phoscon App, can you open the browser developer tools  via F12.
In dev tools click the console tab and reload the page.

Perhaps a Javascript ex exception is shown in red?

PremiumUsername, 2019-02-22:
I think it was chrome's fault somehow. After having deleted all data for phoscon in chrome://settings/siteData things started to work again. Sorry for the false alarm.

manup, 2019-02-22:
Some data is cached in the browser, but of course it should bring the system down.
If it happens again please check for Javascript errors as mentioned above so the issue can be fixed.

closed 2019-06-29 by stale[bot]