dresden-elektronik/deconz-rest-plugin #1291: [Request device support] Elko thermostat

Gaute64Bit, 2019-02-22:
ELKO Thermostat
http://proff.elko.no/elko-smart-home-super-termostat/esh-plus-super-tr-rf-ph-article2629-1779.html 

Need support for reading local temp and setting Heating (and cooling?) Setpoint.

[image]

[image]

[image]

[image]

Contact me if more infor is required.

epiespen, 2019-06-10:
Yes, a big +1 here. Kinda dissappointed there were no support for this.

t112013, 2019-09-25:
oh yes please. this is very much needed :D

dreuf83, 2019-10-23:
Making a +1 on this and hope that someone with the knowledge sees it :)

jyrki69, 2019-11-05:
+1
Would really appreciate support for Elko Thermostat. Let me know if I can help in any way.

epiespen, 2019-11-05:
@ma-ca Is this what you need in reference to #2009 ?

[image]

[image]

t112013, 2019-11-06:
Plsss someone be kind and fix it :)

eidsaa, 2019-11-19:
Hi

There are several more vendor specific attributes, such as floor temperature, child lock, heating on/off etc. In the link below there is a list showing vendor specific attributes. 

https://github.com/prj84/com.Elko/blob/master/drivers/ESHSUPERTR/ST_Code/Elko%20Thermostat%20-%20Vendor%20Specific%20attributes.txt

ricmik, 2019-11-19:
@eidsaa Maybe Elko will be helpful and give you a specification for the vendor specific attributes if you contact them?

eidsaa, 2019-11-19:
@ricmik Yes, it could be. The linked text file list both clusters and encoding. I've implemented floor temp, room temp, child lock and heating on/off in my local ZCL.xml-file. This gives me ability to read and set values from deconz-gui, but not through the REST

corvy, 2019-11-21:
What needs to be done to get this accessible via REST?

epiespen, 2019-11-22:
This is released now, right? How do I even add thermostats in Phoscon?

corvy, 2019-11-22:
It will probably only work in the old web interface. But I do not get it there even if I have it added and it shows in VNC. Have also tried reading the node info to see if that works but I see nothing.

corvy, 2019-11-22:
Phoscon is closed source development.... Sadly.

eidsaa, 2019-11-26:
> What needs to be done to get this accessible via REST?

After the last update, its possible to read and set name, setpoint, etc. through REST.

epiespen, 2019-11-26:
Ok, so there's still no simple way to get this thermostat working in Home Assistant?

eidsaa, 2019-11-26:
I dont think so, Im using Node-RED to send messages and commands between deconz and domoticz.

corvy, 2019-11-27:
It should be possible to get it in Home Assistant but you would need to use REST API to get it, and to set values etc. I just polled with REST now and get the thermostat like this: 

> {
>   "config": {
>     "heatsetpoint": null,
>     "offset": 0,
>     "on": true,
>     "reachable": false,
>     "scheduler": null,
>     "scheduleron": null
>   },
>   "ep": 1,
>   "etag": "ee228d7305c3644d519311c652db9033",
>   "manufacturername": "Heiman",
>   "modelid": "Super TR",
>   "name": "Thermostat 25",
>   "state": {
>     "lastupdated": "2019-11-21T05:55:05",
>     "on": false,
>     "temperature": 1950
>   },
>   "type": "ZHAThermostat",
>   "uniqueid": "00:0d:6f:00:0f:41:45:3a-01-0201"
> }

To find it just connect over a REST client like this: 

> curl -X GET -i 'http://<ip>:40850/api/<restpw>/sensors/<id>'

Just omit the <id> part to list all sensors, then find the ID for this one. You can poll and set states etc. More information on REST here: 
https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

I might write a guide to add it to HA in a week or two... :)

ricmik, 2019-11-27:
How about creating a request to add support for thermostats in the DeCONZ integration on the home assistant repository?

corvy, 2019-11-30:
Yes I agree @ricmik but I have not investigated that option right now. 

Just put the thermostat into HA with REST API:

[image]

Configuration in configuration.yaml:

> sensor:
>   - platform: rest
>     resource: http://<ip>:40850/api/<key>/sensors/<id>
>     name: "Bathroom Termostat"

Create a new sensor section or append to your existing. Replace IP and key and sensor number with your setup. Remember you must setup deconz REST API first: https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

dreuf83, 2020-01-02:
It looks like the only temperature reading the thermostat is sending is the airtemp. I have my thermostat outside of the bathroom with a floor tempsensor. Is it possible to include it to read the floortemp? 

0x409 (encoding:29 value: <verified floor temperature sensor measurement>
0x403 (encoding:30, value: <verified 00=airsensor, 01=floorsensor, 03=gulv vakt (max temp on the floor>
0x408 (encoding:21 value: floating values ex: 001a, 01a9, 01dd, 0000, 0087 <- Average power consumption last 10 min.
0x415 (floating encoding:10, value: <verified heating active/inactive> 00=idle 01=heating

prj84, 2020-01-06:
**LocalTemp** attribute is always the Air measurement on this thermostat

**0x409** attribute is always the floor temp

**0x403** indicates witch sensor that is programed to be used against the setpoint temp on the thermostat itself.

I use the 0x403 attribute in a app I wrote for Athom Homey to show correct measured temp in the app,

**0x415** attribute is just indicating if the heating is active or not. Not possible to override this (Of my knowing)

gbbirkisson, 2020-01-10:
Hi,
I am not able to see the sensor in the REST API when I add the thermostat to the Deconz. I am wondering if you guys know what I am doing wrong.

Rest API Info:
`curl core-deconz:40850/api/<key>/config | jq -r '.apiversion'` -> `1.16.0`
`curl core-deconz:40850/api/<key>/config | jq -r '.swversion'` -> `2.5.72`

I add the thermostat by going to the Phoscon UI and doing add sensor and then make the Thermostat rejoin the Zigbee network. I can see the thermostat in the Deconz VNC view but it does not show up when I call: `curl core-deconz:40850/api/<key>/sensors`

Can someone point to what I am doing wrong? @corvy are those the steps you followed to make it show up in the API?

chairman2s, 2020-04-28:
Just got mine installed yesterday :) I am running HA and deconz. I added it and it appeared immediately in HA. But it only shows the air temp. not the floor temp. And I think there is some trouble setting the temperature and so on... So I vote for better support... What firmware version are you guys running? Mine has 1.2.5R

oywino, 2020-04-28:
> Just put the thermostat into HA with REST API:
> Configuration in configuration.yaml:
> 
> > sensor:
> > 
> > * platform: rest
> >   resource: http://:40850/api//sensors/

I can see the ELKO Thermostat in the deConz VNC interface, but I cannot see it in the REST API or in Phoscon or in HA (after updating the sensor: section).  Can you please tell me how to get in touch with this "bugger" ?

kojax-net, 2020-05-10:
> @ricmik Yes, it could be. The linked text file list both clusters and encoding. I've implemented floor temp, room temp, child lock and heating on/off in my local ZCL.xml-file. This gives me ability to read and set values from deconz-gui, but not through the REST

How did you implement these attributes? When I add these to zcl/general.xml under cluster 0201 I only get unsupported attribute.

```
<!-- ELKO manufacturer specific -->
<attribute-set id="0x400" description="ELKO">
    <attribute id="0x403" name="Sensing" type="enum8" default="0" access="r" required="o" mfcode="0x1002">
        <value name="Local temperature mode" value="0"></value>
        <value name="Floor temperature mode" value="1"></value>
        <value name="Floor max temperature mode" value="3"></value>
    </attribute>
    <attribute id="0x0409" name="Floor Temperature" type="s16" range="0x954d,0x7fff" access="r" required="o" mfcode="0x1002">
    </attribute>
    <attribute id="0x0415" name="Thermostat Running Mode" type="bool" default="0" access="r" required="o" mfcode="0x1002">
        <value name="Idle" value="0"></value>
        <value name="Heating" value="1"></value>
    </attribute>
</attribute-set>
```

Offpiste84, 2020-05-23:
How to get rele status?

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

@Offpiste84 If you have a specific question, please open another issue or ask it on Discord :)

closed 2020-06-05 by Mimiix

corvy, 2020-08-05:
Could we open this issue again to add the floor sensor temperature as-well? 

From a developer of Homey I found that the floor sensor is attributt 0x409, but this does not show in REST. Local Temperature is allways the air temperature. Would that be possible? @Mimiix

Mimiix, 2020-08-06:
@corvy I'd rather have a new device request for that.

oywino, 2020-08-06:
> @corvy I'd rather have a new device request for that.

Why open yet another thread instead of having everything in one place?
It becomes impossible to gather all neccessary information when the discussion is split up into many independant threads.
What's the point?
This thread has an excellent headline - let's leave it alive until the problem is solved, please!

Mimiix, 2020-08-06:
@oywino I have to disagree. How i understand it, it is a floor sensor and slightly different. The title on this issue only suggests the thermostat.

As this is closed already, it doesn't make sense to re-open it because it doesn't comply no more with older changelogs. Having a new issue makes more sense. 

Grouping stuff is easier then ungrouping in the end. And: What is the difference :)?

corvy, 2020-08-06:
Ok great, I will create a new on.

oywino, 2021-03-23:
My ELKO Super TR Thermostat keeps generating these log entries:
```
21:16:16:949 [INFO] - No button map for: Super TR, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0201, command: 0x0A, payload: 010421D007, zclSeq: 225      
21:16:16:949 ZCL attribute report 0x000D6F000E9DC646 for cluster: 0x0201, ep: 0x01, frame control: 0x08, mfcode: 0x0000                                      
21:16:18:957 [INFO] - No button map for: Super TR, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0201, command: 0x0A, payload: 170428E4, zclSeq: 226        
21:16:18:957 ZCL attribute report 0x000D6F000E9DC646 for cluster: 0x0201, ep: 0x01, frame control: 0x08, mfcode: 0x0000
```

Does this mean that there is a mismatch somewhere in the button map file for this device?