dresden-elektronik/deconz-rest-plugin #1292: No device can be controlled any more

l-mb, 2019-02-22:
Hi, all my devices - Hue, Dresden Elektronik FLS, Osram Switches, Amelech - are non-functional.

The web app still shows them, but I cannot change their state, and changing the state manually (say, via pressing the button on the Osram power plug) no longer is reflected in the UI either. None of my ZigBee sensors report anything. In short, everything ZigBee is currently down.

I'm using the RaspBee on Stretch, firmware 2.05.59 (but this started happening with .58).

I have checked the advanced network page introduced via .59, but my panid and network key remain unchanged. I was hoping that'd fix things, but it didn't.

I have tried resetting some devices to force a reconnect, but they don't show up even when scanning for devices.

The deCONZ logs do show "auto connect com /dev/ttyAMA0" "Serial com connected", "Device firmware version 0x262F0500", so I'm assuming that is working and it's running on the right port.

*However*, the ZigBee channel is shown as 0. I'm pretty sure my ZigBee network is on channel 15. Changing this manually, or via reloading an older configuration via the alt-advanced panel, doesn't change this permanently - it reverts back to channel 0.

On the general page, I also get 
`
Vendor dresden elektronik
Product RaspBee
Version 2.05.59 / 2/14/2019
Firmware Not connected
The version is up to date.`

Is that related?

manup, 2019-02-22:
>_However_, the ZigBee channel is shown as 0. I'm pretty sure my ZigBee network is on channel 15. Changing this manually, or via reloading an older configuration via the alt-advanced panel, doesn't change this permanently - it reverts back to channel 0.

This basically says the Zigbee module is  not in operational state.
Do you have access to the deCONZ GUI? We should check the Network Settings.

Also can you check that there are not multiple instances of deCONZ running:

    ps ax | grep deCONZ

l-mb, 2019-02-22:
There are no multiple instances of deCONZ running, and fuser also only shows the one process.

In the network settings, it does show the right PAN id, network key, security mode is "High - No Master but TC Link Key", NWK Update ID 2, channel mask only ticks 15, and NWK Ext PAN ID / APS Use Ext PAN ID are 0x0.

But the state remains stuck at "Joining ..." for several minutes now and doesn't seem to be changing.

manup, 2019-02-22:
Further parameters should be:

* Coordinator
* Mac Address 0x00212effff******
* TC Address = Mac Address

Not sure why it is stuck on rejoining.

To power cycle the module, close deCONZ first and then execute:

    sudo GCFFlasher_internal -r

l-mb, 2019-02-22:
MAC/TC address are right, but the Device Type shows as "Router", not coordinator.

I tried setting it to "Coordinator" and saving, but it reverted back to "Router". A disconnect/reconnect made the new setting show up. (No functional changes though.)

However, if I use the power cycle -r again, it reverts back to router. (All other devices also have "Router" or "End Device", there's no coordinator apparently.)

Is this perhaps a function of the security mode?

manup, 2019-02-22:
Ah ok, can you try the following firmware, it has a fix which should prevent this very problem.

https://www.dresden-elektronik.de/rpi/deconz-firmware/deCONZ_Rpi_0x26300500.bin.GCF

    sudo GCFFlasher_internal -f deCONZ_Rpi_0x26300500.bin.GCF

l-mb, 2019-02-22:
Thanks for the fast feedback, I've flashed the firmware successfully.

Unfortunately, after a device reset, the state reverts to "router". The displayed channel in the web tool also reverts back to channel 0 as before.

Is there anything I can further help to debug this?

FWIW, I also get these in the logs:
15:43:08:737 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 1, node: 0x0000
15:43:08:772 Master: read param with arg 0x19
15:43:08:829 Current channel 0
15:43:08:862 CTRL got nwk update id 2
15:43:08:873 CTRL ANT_CTRL 0x03
15:43:08:958 Device protocol version: 0x0109
15:43:09:006 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 1, node: 0x0000
15:43:09:041 Master: read param with arg 0x19
15:43:09:097 Current channel 0
15:43:09:130 CTRL got nwk update id 2
15:43:09:141 CTRL ANT_CTRL 0x03

and whenever I want to do something with one of the lights/plugs in the UI:
15:43:22:240 API error 901, /lights/3, Internal error, 951
15:43:23:345 API error 901, /lights/4, Internal error, 951

l-mb, 2019-02-22:
Oh! It's working now. Apparently I was too impatient - I didn't watch longer than a few minutes, but now I came back from a break and my devices have appeared again.

Thanks!

johannes-mueller, 2019-03-18:
Similar problem:

I've been running a raspbee on for like a year now without problems. Yesterday the SD of my Raspi crashed. Fortunately I had a backup. However after reinstalling on a new SD-Card, deconz obviously fails to reconnect to the raspbee again.
[image]

When I click "Update Firmware" a progress bar appears and slowly progresses. After reaching 100% it tries to reconnect and the "Update Firmware" reappears.

I tried
'''
pi@cerbo:~ $ sudo GCFFlasher_internal -r
GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10       
no device specified use RaspBee (/dev/ttyAMA0)                                 
reset target SUCCESS
pi@cerbo:~ $
'''

Should I try to update the firmware via GCFFlasher_internal? Which firmware should I use?

manup, 2019-03-18:
Is the serial port configured?

https://github.com/dresden-elektronik/deconz-rest-plugin#software-requirements

johannes-mueller, 2019-03-18:
That was a hint. Thanks. Now I can connect to the raspbee again. All the lights and switches and devices show up. However I cannot switch them. When I try to switch them in the webapp (or in hass) they don't change the state and the state in the UI changes back after a view seconds.

The GUI app (which I actually have not used before) is also behaving strangely. It keeps trying to connect and obviously fails.

Thanks so far.

manup, 2019-03-19:
Please check the Zigbee configuration page as described here:

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

The backup should contain all previously Zigbee configuration parameters. Watch out for changed Channel, PANID or network keys.

johannes-mueller, 2019-03-19:
Now we're getting there. I misread the warning `Important: /boot/cmdline must not contain console=serial0`. Shame on me. Now all the zigbee stuff is working again, the firmware update suceeded.

Thanks for your help.

johannes-mueller, 2019-04-23:
Now I am having similar problems again. While I was on vacation the connection to the RasBee dropped and it won't reconnect.

I updated to 2.05.64, the firmware upgrade succeeds according to the Phoscon app. I have some doubts about that, as the deCONZ program fails to connect.

I also tried with GCFFlasher_internal:
'''
pi@cerbo:~ $ sudo GCFFlasher_internal -l
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyS0       | 0x0000 | 0x0000  |            | RaspBee 
pi@cerbo:~ $ sudo GCFFlasher_internal -r
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device RaspBee (RaspBee)

622: Error: failed to open device, status: Permission error while locking the device (2)
pi@cerbo:~ $ 
'''

manup, 2019-04-23:
>622: Error: failed to open device, status: Permission error while locking the device (2)

This looks like deCONZ is still running or something tries to access the serial port?

johannes-mueller, 2019-04-23:
Ah ok sure. That's why I couldn't connect to it. So my assumption that the connection to the RaspBee dropped was wrong, apparently. Still all the switches and lights are "unavailable". The lights have power, they are actually on.

Any ideas how to diagnose that.

deconz version 2.05.64; firmware 26330500

manup, 2019-04-23:
>Any ideas how to diagnose that.

To start with please check the *Zigbee configuration* page if any parameters like channel or PANID have changed, if they did you can revert to a previous working configuration.

Here is how the page can be opened:
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

johannes-mueller, 2019-04-23:
I compared the configuration in the deCONZ app with the zbconf table in an old zll.db, that I had backuped from a working setup. Indeed the PANID and the network key had changed. I reverted both to the old version.

Two more things I suspect:
* In deCONZ it said "Security Mode: No Security"; in zll.db it said `"securityMode":3`
* In deCONZ the NWK Update ID is 4: in zll.db it's `"nwkUpdateId":0`

I tried out the other Security Modes and changed the NWK Update ID to 0.

Still the same behavior. No light available; sensors "unreachable"

manup, 2019-04-23:
Looks very much like configuration got messed up somehow.
You don't need to fix this manually, the Load button on the Zigbee configuration page should take care of all this and fix invalid configurations.

Note Usually the valid configuration is the one for which many entries exist.

johannes-mueller, 2019-04-23:
Ok. Works again. 

At first loading the backup on the zigbee configuration page failed. Then moved the config in `/home/pi/.local/share/dresden-elektronik` away. Then I rebooted the RasPi and then loading the backup succeeded. Now everything seems up and running again. Phew.

 Thanks so much for your help.

1968-Larry, 2019-04-23:
I also got problems with not connected, it started earlier this morning I think. I tried to upgrade everything, manually flashed the RaspBee with no success. I did @johannes-mueller trick and moved the deCONZ folder and started everything again. Loaded the backup, deCONZ now shows 41 nodes but only the Coordinator show last seen information and the network seems dead. I reset the raspberry and tried to join a new bulb and it worked. I tried again to restore the backup. And now it works. I got all files backed up manually if someone is interested to look at.

closed 2019-08-28 by stale[bot]