dresden-elektronik/deconz-rest-plugin #1294: [Request device support] - Develco Motion Sensor Mini

Hagenisse, 2019-02-23:
I'd like support for this device in deCONZ, if possible?
In Phoscon GW, when trying to connect the sensor as type "Other", it says "Failed to connect". It appears in deCONZ, but only with one node cirle:

[image]

I guess that means I won't be able to provide any cluster info?

The node info looks like this:
[image]

Product details from the manufacturer:
https://www.develcoproducts.com/products/sensors-and-alarms/motion-sensor-mini/

Ronoon, 2019-02-23:
delete the device in DeConz and then reconnect it to deconz to you can follow the procedure as written in 

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

Without  it it will mostly not be possible to give you the requestet support
Since the device has the HA1.2 profile you should be able to fullfill the answers to the requestet device support sheet

ebaauw, 2019-02-23:
Or try and select _Read Simple Descriptor(s)_ from the left circle, while pressing the dimmer's buttons, to make sure it is awake.  When the node blinks blue, deCONZ sends the request.  If it blinks red afterwards, the device could not be reached, probably because it's asleep with the radio receiver off.

Hagenisse, 2019-02-24:
Thanks for your replies!
And I'm sorry, I somehow managed to upload the node info for the wrong device! (A working dimmer) 

Please, let me try again. This is a motion sensor. After deleting it from deCONZ, opening the network (permit join), resetting the sensor, keeping it awake and let deCONZ find it (did not use Phoscon this time), I ended up with both node circles:
[image]

Here is the CORRECT node info for this sensor:
[image]

Also, I tried ebaauw's suggestion and hit the _Read Simple Descriptor(s)_ a few times. Wich led to this info on the right node circle:
[image]

However, I can't seem to get any meaningful **Cluster Info** when clicking on the lines. If I click on the line "0000 Basic (22)" for instance, it gives me this:

[image]

But there seems to be _some_ info on a couple of lines. Here's from "0500 IASZone (5)":
[image]

And from "0019 OTAU (8)":
[image]

Does this help?
Is there anything else I can do or information somewhere I can provide?
Or is this a lost case as it still seems impossible to get meaningful cluster information?

Ronoon, 2019-02-24:
Well that should help so @manup  can look for a fault.
Got a similar problem and had to reset the sensor more times to default and then integrating it once again. So maybe you should try it once again to get a better result. 
In my case it was helpful and i got it once again to responding.

manup, 2019-06-25:
We still need the attribute data from the Basic cluster like model identifier.

>However, I can't seem to get any meaningful Cluster Info when clicking on the lines. If I click on the line "0000 Basic (22)" for instance, it gives me this:

If this happens please restart deCONZ and try again, there seems to be a bug where the cluster widget is occasionally incomplete.

jukmanty, 2019-10-12:
I have similar device:

[image]

Link to the technical manual of the device:

https://www.develcoproducts.com/media/4314/moszb-130-140-technical-manual-motion-sensor.pdf

closed 2020-02-16 by stale[bot]

jukmanty, 2020-02-23:
I managed to connect MOSZB130 sensor using V2_05_73 beta version: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2011#issuecomment-590077023

PrimaryInc, 2021-04-08:
Is it possible to reopen this thread or should I create a new "issue"/"request"

Mimiix, 2021-04-08:
@PrimaryInc can you create a new one and add this issue as a reference?

PrimaryInc, 2021-04-09:
> 
> 
> @PrimaryInc can you create a new one and add this issue as a reference?

Yes, will do :)