dresden-elektronik/deconz-rest-plugin #1297: Issues with Xiaomi sensors and buttons

christmasjumper, 2019-02-25:
Hi,

I seem to be having a really difficult time getting any of the Xiaomi sensors and buttons to reliably work after upgrading to 2.05.59.

My door sensors are missing open / close events, and the button we use as our doorbell is not triggering event ID's in home assistant at all, despite deleting and readding the button.

I am not sure what logging or config information would be useful at this stage, but really do need some help please.

P.S I also have Ikea lights connected, they've been working fine and the newest software has removed the flickering light issue - thank you!

Oliviakrkk, 2019-02-25:
+1 this happens to xiaomi buttons, aqara wall buttons (battery powered) and Door sensors

christmasjumper, 2019-02-25:
I have been looking in the home assistant debug logs and from what I can see the button events are showing, but they don't have the right names or ID's i'd expect. I am not sure if this is something Deconz or HAS side to be honest (I'll also post there)

2019-02-25 12:41:49 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"e":"changed","id":"27","r":"sensors","state":{"dark":true,"lastupdated":"2019-02-25T12:41:49","presence":true},"t":"event","uniqueid":"00:0b:57:ff:fe:92:1e:87-01-0006"}
2019-02-25 12:41:49 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"e":"changed","id":"27","r":"sensors","state":{"dark":true,"lastupdated":"2019-02-25T12:41:49","presence":true},"t":"event","uniqueid":"00:0b:57:ff:fe:92:1e:87-01-0006"}

CopyCat73, 2019-02-25:
Something seems to have changed in a recent version. Initially I installed 2.05.52 under ubuntu and was able to add a bunch of xiaomi contact sensors and buttons. Then I moved to windows, the default version was a lot lower (2.05.20). I did a firmware upgrade and after that not a single xiaomi device can be found anymore. Plugging the conbee back in the ubuntu installation also gives no results anymore so I assume something is wrong in the stick firmware?

heitec11, 2019-02-27:
My Xiaomi door sensor is also not working with 2.05.59

christmasjumper, 2019-03-01:
Any update on this? The buttons not working is the biggest pain as I have a few them set up for various actions around the house *headdesk*

Oliviakrkk, 2019-03-04:
The issue is the buttons sometimes work sometimes not. I can see that if buttons weren't used for some time the first action will not work. The second try will be succesffull and event will be sent to home assistant. 

It did some tests on one of the buttons. Here are logs from the first click:

19:43:24:976 no button map for: lumi.sensor_magnet ep: 0x01 cl: 0x0000 cmd: 0x0A pl[0]: 002
19:43:24:976 ZCL attribute report 0x00158D000171E9E0 for cluster 0x0000, ep 0x01
19:43:24:977 0x00158D000171E9E0 extract Xiaomi special attribute 0xFF02
19:43:24:977    64 on/off 0
19:43:24:977    01 battery 2985 (0x0BA9)
19:43:24:977    03 unsupported index (data type 0x21)
19:43:24:977    04 unsupported index (data type 0x24)
19:43:24:977    05 unsupported index (data type 0x21)
19:43:24:977    06 unsupported index (data type 0x20)
19:43:32:720 0x00158D000171E9E0 error APSDE-DATA.confirm: 0xF0 on task
19:43:32:720 APS-DATA.confirm id: 76 status: transaction expired (8 seconds)
19:43:34:611 0x00178801021D61FE onOff 1 --> 0

19:45:48:271 0x00178801029AA1D7 error APSDE-DATA.confirm: 0xE9 on task
19:45:56:693 button 1002 Normal release
19:45:56:695 ZCL attribute report 0x00158D00018B4A84 for cluster 0x0006, ep 0x01
19:45:57:207 no button map for: lumi.weather ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000
19:45:57:207 ZCL attribute report 0x00158D0002725634 for cluster 0x0402, ep 0x01
19:45:57:238 no button map for: lumi.weather ep: 0x01 cl: 0x0405 cmd: 0x0A pl[0]: 000
19:45:57:238 ZCL attribute report 0x00158D0002725634 for cluster 0x0405, ep 0x01
19:45:57:269 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000
19:45:57:270 ZCL attribute report 0x00158D0002725634 for cluster 0x0403, ep 0x01
19:45:59:300    0x00178801021D61FE force poll (2)
19:46:00:357 0x00178801029AA1D7 error APSDE-DATA.confirm: 0xA7 on task
19:46:16:877 0x00178801029AA1D7 error APSDE-DATA.confirm: 0xE9 on task
19:46:17:695 button 1000 Normal press
19:46:17:698 ZCL attribute report 0x00158D00018B4A84 for cluster 0x0006, ep 0x01
19:46:18:690 button 1001 Hold
19:46:19:892 Set sensor check interval to 1000 milliseconds
19:46:28:484 button 1000 Normal press
19:46:28:484 ZCL attribute report 0x00158D00018B4A84 for cluster 0x0006, ep 0x01
19:46:28:675 button 1002 Normal release
19:46:28:677 ZCL attribute report 0x00158D00018B4A84 for cluster 0x0006, ep 0x01
19:46:28:850    0x00178801021D61FE force poll (2)
19:46:30:098 Set sensor check interval to 1000 milliseconds
19:46:30:916 button 1000 Normal press
19:46:30:917 ZCL attribute report 0x00158D00018B4A84 for cluster 0x0006, ep 0x01
19:46:31:094 button 1002 Normal release
19:46:31:095 ZCL attribute report 0x00158D00018B4A84 for cluster 0x0006, ep 0x01
19:46:31:251    0x00178801021D61FE force poll (2)
19:46:32:691 Set sensor check interval to 1000 milliseconds

Oliviakrkk, 2019-03-04:
I downgraded to 2.05.52...Will see if that helps.

azrarel783, 2019-03-06:
@Oliviakrkk Could you resolve the problem by downgrading? Any news on this issue? I have version 2.05.57 running and I lost one temperature/humidity sensor from Xiaomi and I'm not able to reconnect this one, nor others ...

Oliviakrkk, 2019-03-07:
Yes. I've been testing for the last two days. 

2.05.52 with home assistant 0.86 works flawlessly.

I will update home assistant this evening and we will see whether issue is with new deconz or new home assistant.

azrarel783, 2019-03-07:
Ok, sounds good. How did you do the downgrad? Just downloaded the version 2.05.52.deb and reinstalled it, or did you first uninstall the newer version?

I hope that the issue is not related to home assistant. How did you add your devices? I used the phoscon app setting up all my devices and added then the integration of deConz into home assistant. And phoscon is also the place where I have the issue with adding the Xiaomi senonrs.

Oliviakrkk, 2019-03-07:
I have docker. So downgrade is very easy (container swap).

I added all my sensors via phoscon. My deconz is more than 3 months old so I was adding devices over time over many versions. 

 I had some issues with adding xiaomi until I added more relays to my network (relays in my case are hue lights which are connected to 230V all the time).

azrarel783, 2019-03-07:
Ok, I used the SD card image for the Rpi. I just checked the Dresden Elektronik webpage, but I was not able to find it anymore. So guess I will change also to Docker being afraid that I have to setup all my devices from scratch -.-

Oliviakrkk, 2019-03-07:
I think docker is the easiest to maintain and backup :).

christmasjumper, 2019-03-07:
> Ok, sounds good. How did you do the downgrad? Just downloaded the version 2.05.52.deb and reinstalled it, or did you first uninstall the newer version?
>
> I hope that the issue is not related to home [...]

Downgrading is generally quite easy, just run the install command but specify the old version....

Make sure you take a backup through the webgui first.

azrarel783, 2019-03-07:
@christmasjumper Thank you for the information, but I like Docker and think I will change to it.

Oliviakrkk, 2019-03-08:
Ewerything works with 2.05.52 and  home assistant 0.88. Looks like xiaomi issues are connected with deconz 2.05.59

azrarel783, 2019-03-08:
Nice, thank you @Oliviakrkk for the information.

Oliviakrkk, 2019-03-10:
Hi @manup ..is this a known issue?  It is definitely problem with 2.05.59...

azrarel783, 2019-03-12:
I'm also back to 2.05.52 and was able to add new Xiaomi temperature/humidity sensors. But I had to try it more than one time to get them show up in Phoscon app.

Oliviakrkk, 2019-03-12:
I updated yesterday to.. 60. Will see over the next few days how it works (so far good :))

azrarel783, 2019-03-20:
@Oliviakrkk How is the experience with .60 so far? Is it running stable?

Oliviakrkk, 2019-03-20:
I think I still have issues with xiaomi buttons :/.

Hue dimmers work great, no delay. With xiaomi I need to push buttons multiple times... Looks like they lose connection when not used for some time or something like that.  When button starts working it will work flawlessly for some time and then again will stop working. Then I need to press it multiple times and then it will again work for some time. 

I need to have free evening to try to run debug on my deconz...

brettyj, 2019-07-09:
I'm seeing the same thing with my xiaomi button on 2.05.66 - if it hasn't been pushed in a while the first press doesn't register, need to press a 2nd time for it to register.

My door sensors seem to be reliable though *touch wood*

closed 2019-11-13 by stale[bot]