dresden-elektronik/deconz-rest-plugin #1298: REQUEST: Sensor/Gerät hinzufügen via Phoscon Smartphone-Website

kugelkopf123, 2019-02-25:
Es wäre super wenn Sensoren bzw. Geräte auch in der Mobile-Version der Phoscon - Website hinzufügen könnte. Gerade bei den Xiaomi/Aqara Sensoren ergibt es Sinn, diese direkt an der Stelle ins Netz zu bringen, an der sie später auch ihren Dienst verrichten sollen. Sie sind ja leider nicht so kontaktfreudig bei neuen/anderen Geräten in wechselnder Umgebung.

Ronoon, 2019-02-25:
First of all: This is an english spoken/written Website  so please use the english language

Second: no use by installing the sensors etc by the Point of Installation cause there are to many factors that are hindering the correct Installation far away from the Phoscon Gateway/Stick ( you never know if the Connection is ok or not if your to far away, so for the first Installation do this near the Gateway)
So if you have install Problems it could be by the interference of other WiFi components or to many walls between the Gateway etc.
Hope this will help so please do the install near the gateway

kugelkopf123, 2019-02-26:
I'm sorry, I thought, because it is a German product ...
This is nonsense and has nothing to do with the gateway. The Xiaomi / Aqara sensors can also log in to other routers (such as power sockets, light bulbs etc...) on the Zigbee Network and should work there as good as near the gateway. Unfortunately, as they have problems to make new connections (what is also mentioned here on github) to other nodes, it makes sense to add the sensors in the environment in which they will work later.
Regardless, it is a request if it would be possible to integrate this "Add Button" into the Mobile Website. No matter whats with my Aqara Sensors ;).

Larsn1, 2019-02-27:
I have already added sensors in mobile view. Have you tried panorama view? In safari there is a plus button if you turn the device.

Wyphorn, 2019-02-27:
Oh yeah, I really rarely use my desktop pc. It's really oldschool to add a hue dimmer on a real pc. It should be absolutely possible on my mobile.
Although requesting "dektopwebsite" on the mobile will fail. I am using all4hue to add my normal hue devices. But for xiaomi, I have to start my desktop..
So: Please let us add sensors on mobile too!

manup, 2019-02-28:
I can't provide an ETA yet but at least the generic "Add sensor/switch" buttons for mobiles are on the todo list.

Note on tablets these should already work.

closed 2019-07-05 by stale[bot]