dresden-elektronik/deconz-rest-plugin #1300: Restore raspbee backup in conbee

Wyphorn, 2019-02-26:
Hi, I switched from RaspBee to Conbee and tried to load my raspbee backup in conbee.
But the startup routine loops with these logs:
`Feb 26 21:32:28 server systemd[1]: Started deCONZ: ZigBee gateway -- Initialisation.
Feb 26 21:32:28 server systemd[1]: Started ACPI event daemon.
Feb 26 21:32:28 server systemd[1]: deconz.service: Service hold-off time over, scheduling restart.
Feb 26 21:32:28 server systemd[1]: Stopped deCONZ: ZigBee gateway -- REST API.
Feb 26 21:32:28 server systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Feb 26 21:32:28 server systemd[1]: Starting deCONZ: ZigBee gateway -- Initialisation...
Feb 26 21:32:28 server deCONZ[24643]: QXcbConnection: Could not connect to display
Feb 26 21:32:28 server deCONZ-init.sh[24644]: cat: /sys/block/mmcblk0/device/cid: No such file or directory
Feb 26 21:32:28 server deCONZ-init.sh[24644]:  stopping deconz-wifi.service
Feb 26 21:32:28 server systemd[1]: Stopped deCONZ: ZigBee gateway -- WIFI Service.
Feb 26 21:32:28 server  systemd[1]: Reloading.`

Is there a nice way to migrate? I have 30 hue devices, multiple sensors and remotes and all is connected to fhem with its names. I don't want to reset all, learn them to new network and set all up again..

Best

etique57, 2019-02-27:
I'm planning to do the eact same thing (sick of the fragility of the raspberry...) so I'm interested in the answer!

etique57, 2019-03-04:
Any idea or piece of advice?

manup, 2019-03-04:
>Is there a nice way to migrate? 

The simplest way is to load the backup of the old installation (created in the Phoscon App).

However keep in mind that the RaspBee shouldn't be powered afterwards since two clones (via mac address) would fight over the network.

But the above log looks a bit like the ConBee/ deCONZ installation isn't working for some reason.
What kind of  system is this?

Wyphorn, 2019-03-06:
This is an ubuntu debian 16.04 lts 64bit without x11.
After I installed deconz I was able to browse to phoscon webpage.
Then I created the backup of my phoscon running on raspbee (raspberian image of dresden electronics) with x11.
Then I uploaded this backup to deconz with conbee and it was imported, but no lights were displayed. So I restarted the server and deconz was not longer able to start. I always get the error messages.

I tried to reinstall deconz on my server but it is still failing with the provided errors. What files do I have to delete for a full deinstallation?
I would test to run deconz on raspbee headless to before I create the backup.

Wyphorn, 2019-03-06:
So, I tried again to remove all deconz folders, but it's still like this:
`Mar  6 21:28:42 server systemd[1]: Stopped deCONZ: ZigBee gateway -- WIFI Service.`
`Mar  6 21:28:42 server systemd[1]: Reloading.`
`Mar  6 21:28:42 server systemd[1]: Started ACPI event daemon.`
`Mar  6 21:28:42 server systemd[1]: Started deCONZ: ZigBee gateway -- Initialisation.`
`Mar  6 21:28:42 server systemd[1]: deconz.service: Main process exited, code=dumped, status=6/ABRT`
`Mar  6 21:28:42 server systemd[1]: deconz.service: Unit entered failed state.`
`Mar  6 21:28:42 server systemd[1]: deconz.service: Failed with result 'core-dump'.`
`Mar  6 21:28:42 server systemd[1]: Starting deCONZ: ZigBee gateway -- Initialisation...`
**`Mar  6 21:28:42 server deCONZ-init.sh[10754]: cat: /sys/block/mmcblk0/device/cid: No such file or directory`**
`Mar  6 21:28:42 server deCONZ-init.sh[10754]:  stopping deconz-wifi.service`
`Mar  6 21:28:42 server systemd[1]: Stopped deCONZ: ZigBee gateway -- WIFI Service.`
`Mar  6 21:28:42 server systemd[1]: Reloading.`
`Mar  6 21:28:42 server systemd[1]: Started ACPI event daemon.`
`Mar  6 21:28:42 server systemd[1]: Started deCONZ: ZigBee gateway -- Initialisation.`
`Mar  6 21:28:42 server systemd[1]: deconz.service: Service hold-off time over, scheduling restart.`
`Mar  6 21:28:42 server systemd[1]: Starting deCONZ: ZigBee gateway -- Initialisation...`
`Mar  6 21:28:42 server systemd[1]: Stopped deCONZ: ZigBee gateway -- REST API.`
`Mar  6 21:28:42 server systemd[1]: Started deCONZ: ZigBee gateway -- REST API.`
**`Mar  6 21:28:42 server deCONZ[10790]: QXcbConnection: Could not connect to display`**
`Mar  6 21:28:42 server deCONZ-init.sh[10789]: cat: /sys/block/mmcblk0/device/cid: No such file or directory`
`Mar  6 21:28:42 server deCONZ-init.sh[10789]:  stopping deconz-wifi.service`
`Mar  6 21:28:42 server systemd[1]: Stopped deCONZ: ZigBee gateway -- WIFI Service.`
`Mar  6 21:28:42 server systemd[1]: Reloading.`

I think there are two problems. I marked the lines bold. First is a wrong path in the file "/usr/bin/deCONZ-init.sh". There is a line:
`THIS_SDCARD=$(cat /sys/block/mmcblk0/device/cid)`
On my raspbee I have this path. On my ubuntu server I haven't. Do I have to change something there?

Wyphorn, 2019-03-06:
Okay, I found this command for start up in another issue:
`deCONZ --dbg-info=1 --dbg-aps=1 --dbg-zcl=1 --dbg-zdp=1 --http-port=80 -platform minimal`
This one works for me. I was able to start phoscon with that and was able to import my backup. After that I had again the `QXcbConnection: Could not connect to display`in my logs. But I was able to restart deconz with same parameters again.
The command "deCONZ -platform minimal --http-port=81" in the "/lib/systemd/system/deconz.service" fails with the QXcbConnection error. Running the same command in terminal works nice.
I have no idea for this.

Wyphorn, 2019-03-06:
After starting the deconz with imported files, I have the next problem.
On Phoscon landing page, I am able to use the imported network; the conbee itself keeps greyed out.
In the imported network I see all my bulbs, strips, switches and sensors. But I cannot use them. I tried to search for new lights and turn them off for 10 seconds but I still can not use them.
How to "migrate" them to new hardware?

Wyphorn, 2019-03-11:
Is here any news how to use a restored backup?
The problem is still: I can import the raspbee phoscon backup in deconz and I see all devices, rules and scenes, but I cannot use them. The status of the devices is wrong too.

I tried just a new light search, but without any result. I tried to cut off the power of the devices for multiple seconds and after turning on I still can't find them.
What's the provided way to restore a backup on different hardware? The wiki page is very basically: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Backup-&-Restore

Wyphorn, 2019-03-17:
> I'm planning to do the exact same thing (sick of the fragility of the raspberry... moving to x86 proxmox) so I'm interested in the answer!

I will swap back to hue bridge as main zigbee host. The idea of deconz is nice, but it is not that nice to handle and I think they don't know the onus of a smart home server.
 I forced my switches to homematic, so there is no need for zll support for gira or busch jaeger. For my few xiaomi devices, I will stay with conbee.
With the hue bridge, I had never losing switch devices or problems with updates. They have a much better qa department and with hue bridge, I will be able to get firmware updates and have the "correct hue way" of scenes. I use fhem as smart home server and there I had to create the light scenes at the moment, because deconz gone its own way. Just as a small exchange of experiences.

closed 2019-04-02 by Wyphorn