dresden-elektronik/deconz-rest-plugin #1303: Suddenly lost all but a few connections

ACrazyConcept, 2019-02-27:
When I came home from work almost all lights were greyed out in home assistant. Initially I rebooted the deconz raspberry, and checked in the deconz ui and saw almost no connections. Then I did an upgrade to latest version. still the same.
I have the raspbee on a dedicated pi.

ACrazyConcept, 2019-02-28:
I tried flashing the raspbee now. No change. Tried the increase NWK update id a handful of times. No change.
I don't have any more ideas myself.

manup, 2019-02-28:
Can you check the *Zigbee configuration* page introduced in 2.05.59

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

Have a close look to PANID, Channel and Network key, if these have changed recently, revert back to a previous configuration.

ACrazyConcept, 2019-02-28:
> Can you check the _Zigbee configuration_ page introduced in 2.05.59
> 
> https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59
> 
> Have a close look to PANID, Channel and Network key, if these have changed recently, revert back to a previous configuration.

Thanks @manup 
The panid has not changed since I bought the raspbee.
As you see I have upped the NWK id manually a couple of times.
But I also see that the network ID has changed when I upgraded to latest deCONZ. I do not know what that means?

[image]

closed 2019-02-28 by ACrazyConcept

manup, 2019-02-28:
Yes looks like the issue appeared after the 2.05.54 to 2.05.59 update.
It's a known firmware issue which will be fixed with the next firmware update.