dresden-elektronik/deconz-rest-plugin #1304: Ikea ota upgrade on windows

PrimaryInc, 2019-02-27:
Hi guys,

I've been trying for a while to fine a way to upgrade my ikea trådfri devices via deconz using ikea's ota files.
Found a few guides that shows how this is to work on rasperry and similar, but nothing for windows.

I where able to download the ota files by looking inside ikea's json file, so I have a bunch of ota.ota.signed files. It seems I can load the files, but in every guide I can find it seems the ota files has been unpacked or converted to .zigbee files.

Anyhow, I'm unable to upgrade my bulbs, sensors and buttons, hope someone has input to help med out.

Philje123, 2019-03-02:
Have you installed deconz on windows?

You do it via that.

Download the firmware from here (open the JSON file and look for the product you need the firmware for. There is then a download link for the firmware):

http://fw.ota.homesmart.ikea.net/feed/version_info.json

I've updated 3x wireless dimmers successfully.

In deconz click on plugins at the top and choose OTA. In that window find the device you want to update, choose the OTA file at the top and then click update. Power the device off and on and it should start to update. Battery devices take the battery out and back in. It might take a few tries.

PrimaryInc, 2019-03-02:
Hi @Philje123 ,

Yes of course Deconz is installed :)
So if you successfully upgraded via deconz, do you have a guide?

Per my question, Im unsuccessfully so stating that "use deconz to upgrade" dont get med any closer :)
Im asking for assistance to figure out what Im doing wrong.

Example:
How to you download ota files using the json file on windows?
Can you explain how you do it, step by step?

Only guides I can find, is based on linux and not windows.

Philje123, 2019-03-02:
I don't have a guide no.

But it's straight forward.

Like I said. Click on plugins at the top, then OTA (or SOTA or similar), it opens a tab at the bottom. Click on it and there should be a list of all the zigbee devices. Find the one you want to update and choose the OTA file in the box above and then click update.

As for downloading the files if you open the .JSON file it contains all the Tradfri models. Find your bulb, sensor or switch in there and look for the URL to the firmware file to download it. These files can be used directly in deconz to update.

PrimaryInc, 2019-03-02:
Hi @Philje123 ,

Thanks again for you reply.
As stated I've downloaded the OTA files, and I've used Deconz to try updating Ikea devices, but nothing is happening. So it is something that is not working or I'm missing something. 

Thats why I've asking for a guide, so I can verify steps and specific for windows sins I only can find for linux.

Not sure is applicable to deconz on windows, but I can see from the guides I've found that when starting an update there is created .zigbee files. This does not happen in my environment.

Philje123, 2019-03-02:
Hi,

What firmware version is on your conbee?
Mine is on 262F0500

And which windows version are you using?
I'm on Windows 10 64 bit.

Just trying to check for any differences between my setup and yours.

The files I downloaded are as such:

"159701-TRADFRI-wireless-dimmer-1.2.248.ota.ota.signed"

Like I say I have done 3x dimmers. Don't do anything else but choose the firmware file in deconz and click update.

I then pulled the battery out and put it back in (wait 30 seconds or so) and the info in the OTA windows should populate with firmware version etc and the update should start. If it doesn't work then try again a couple of times.

PrimaryInc, 2019-03-08:
Hi @Philje123,

I'm on the latest firmware as well with deconz .59
Windows 2016 Standard - 64Bit

OTA files location:
C:\Program Files (x86)\deCONZ\otau

Files looks like this:
159694-TRADFRI-bulb-ws-gu10-1.2.217.ota.ota.signed

Does it mean that you have no .zigbee files in you otau catalog?
I need to do a new test on a dimmer as well to verify. 

Did you ever use the tab "ota file" in ota upgrade plugin?
Or just the ota update tab?

closed 2019-07-13 by stale[bot]