dresden-elektronik/deconz-rest-plugin #1305: Ubisys S2 firmware update instructions

saurabh984, 2019-02-28:
Is there any instructions on how to do firmware updates using deCONZ for Ubisysy devices?

ebaauw, 2019-02-28:
See #712.

saurabh984, 2019-02-28:
Thanks @ebaauw 

Any idea what’s the difference between rms2 and fms2?

I’m assuming ‘rm’ is the DIN Rail module

ebaauw, 2019-02-28:
I never decoded the full file name, beyond the obvious fields (see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/712#issuecomment-465744443).  I used the _Image Type_ to select the right file, but deCONZ should be able to figure that out itself if you unpack the files to `~/otau`.

saurabh984, 2019-02-28:
Hmm I did extract them in that folder but I still had to go manually select a file. I’m trying it with fms2

Is your dimmer DIN Rail or switch mounted one?

ebaauw, 2019-02-28:
> I’m trying it with fms2

You should see the _Image_ in the STD OTAU plugin panel (press _Query_ when the S2 is selected in the panel).

> Is your dimmer DIN Rail or switch mounted one?

Switch mounted (the non-R version).

closed 2019-02-28 by saurabh984