dresden-elektronik/deconz-rest-plugin #1306: Phoscon = Script bomb :)

patrickjane, 2019-02-28:
I am using Chrome with ScriptSafe Plugin, when opening http://dresden-elektronik.de/pwa/, it seems to get into an endless loop, trying to load more and more scripts, which are however blocked by ScriptSafe.

[image]

As you can see there is 360 scripts blocked, but the number is increasing at insane speed. This leaves Chrome in a stuck/frozen state, and Chrome needs to be killed manually. After re-opening Chrome and restoring tabs, you have to be real quick to close the phoscon tab, otherwise you have to kill it again.

In general (also from security perspective) I would prefer that the phoscon manual directs users towards using their *actual* hostname/IP they should *very well know*, as they have just installed your software on it. Having something from an external website which starts scanning your network, just to give you the IP/hostname if your just finished installation is really *rather unusual*. Although you assure that everything remains within our local networks, this still leaves kind of a bad taste.

BTW: Not meant to issue harsh criticism :) Just installed deCONZ yesterday and I must say I am really happy with it. Keep up the good work 👍

ebaauw, 2019-02-28:
You can load the Phoscon webapp locally, from the hostname/IP you installed deCONZ to.

manup, 2019-02-28:
As ebaauw mentioned you can open the interface directly via ip

    http://<ip-address>:<port>/pwa

The http://dresden-elektronik.de/pwa page is basically useful when the ip address isn't known. There can be many reasons for that, often non tech safe people get an easier entry point, or in larger networks (perhaps with multiple gateways) which are just plugged in and get a random ip address this method becomes handy.

If the gateway is discovered via internet service no scan should be needed and the gateway will pop up right away, this unfortunately doesn't work in all cases.

Note there are not hundreds of scripts loaded :) but the search scans local ip addresses to figure out gateways in the network.

Closing the issue for now, since we the scan mechanism will stay for a while.
The internet discovery service will improve in next releases to fix some issues with mixed IPv4/v6 networks.

closed 2019-02-28 by manup

patrickjane, 2019-03-01:
Thanks for the reply. Please note that I was *not* trying to find a solution, I *do know* I can access phoscon directly as you've mentioned.

I was reporting a bug in your website. I know it doesn't load thousands of scripts, but in conjunction with a scriptblocker the website leads to a loop which screws the browser.

BTW: I get the point of the scanning, and very well understand the idea that its meant for beginner users since deCONZ also targets non-professional users, and I am perfectly fine with it. Although one might argue that someone who manually installs debian packages or even compiles the sources will not be a beginner in the first place :)

manup, 2019-03-01:
>I was reporting a bug in your website. I know it doesn't load thousands of scripts, but in conjunction with a scriptblocker the website leads to a loop which screws the browser.

Hmm indeed, I'm not sure if it is possible to reliably detect if a script/request blocker is active. But it sure would be handy if it is detectable to show a more useful message to the user instead of infinite search.

I'll put it on the todo list to check.

patrickjane, 2019-03-01:
I am using ScriptSafe for a long time now, and never encountered this behaviour on any other side. However, I've seen lots of sites which dont show their content and state you shall deactivate your adblocker. So technically it should be possible. 
But I'm just a backend developer, I don't know much about web development :)