dresden-elektronik/deconz-rest-plugin #1307: Allow /config/export from rest api

zoic21, 2019-02-28:
Hello,

Already congratulations for the product that works really well.

I have a small request, can you allow /config/export from rest api, With that I can auto launch backup of network every.

Thank in advance,

ebaauw, 2019-02-28:
Should already work.  You need to POST to `/api/`_username_`/config/export`.  The API replies with `"success"` and creates backup file `~/.local/share/dresden-elektronik/deCONZ/deCONZ.tar.gz`.

zoic21, 2019-03-01:
Hello,

Thank a lot it's perfectly work (before I made get request not post, it's my mistake sorry)

closed 2019-03-01 by zoic21

ITSSeu, 2020-01-19:
Hello, I am totally sorry, that I reopening this thread.

I am using deCONZ through apt-get install for a few days and now beginn to play around with the api.

deCONZ is running on my RPI 4 with Buster and I want to be able, to run a light cron bash-script for my backup of deCONZ Phosocon APP and api. This thread give's me the only information about receiving the export via an api call with CURL.
 
I want to connect to the api from the same machine as deCONZ is locally running on. For that, I already generate an apikey:
(command: 
curl -k -S -v -H "Content-Type: application/json" --data @apikey_body.json http://localhost:9090 -X "POST /api"

apikey_body.json:
{ "devicetype": "NEWUSERNAME" }
)

When I call with this apikey, I also received 4 files to ~/.local/share/dresden-elektronik/deCONZ:
config.ini  session.default  zcldb.txt  zll.db

I think that the files are okay, but I find no way to import them via api. Only phoscom app get the option to restore a backup file in the .dat filetype. Like the one I can manually download from the phoscom app.

I already compress the files to tar file, also rename the tar file in .dat but the file can not import by phoscom app.

Can sombody help me please. I am very new in deconz, api and curl. Thanks in advanced :)