dresden-elektronik/deconz-rest-plugin #1308: [Request Device Support] Environment Sensor

martynwendon, 2019-02-28:
I'm moving over to deCONZ from another platform and having being using some DIY ZigBee Environment sensors that offer temperature, pressure, humidity and illuminance measurements.

They are available from a SmartThings / Hubitat user - https://community.hubitat.com/t/hubitat-with-homemade-temperature-humidity-pressure-and-light-sensor

Appreciate that these are probably a bit specialist, but they work very well and are built as router devices (as opposed to end devices) with battery backup capability.

I'm new to deCONZ but under Hubitat / SmartThings for example these are pretty easy to support because the basic functionality uses standard ZigBee HA protocol - https://github.com/iharyadi/hubitat/blob/master/Environment%20SensorEx/Environment%20SensorEX.groovy

Hopefully adding support to the deCONZ REST-API wouldn't be too complicated?

[image]

[image]

ebaauw, 2019-02-28:
It's probably enough simply to whitelist the sensor.  That should create a ZHATemperature, a ZHAPressure, a ZHAHumidity, and a ZHALightLevel `/sensors` resource.  Do the _Binary Input_ and _Analog Input_ clusters report anything useful?

martynwendon, 2019-02-28:
Wow thanks for the quick reply :-)

Yes it's actually a very capable device and can be configured in many different ways, for example you can connect your own sensors to the analog / binary inputs.

From memory the on/off controls diagnostic reporting from the device, quite a cool feature as it allows you to see number of attached children, neighbours, dropped packets, etc.

But I think the basics are enough to start with - if battery percentage could be added that would be a bonus (it has an optional charging circuit so you can attach an 18650 cell for battery backup capability).  Battery percentage is reported from 0 - 200 though so it's not an actual percentage (it is divided by 2 in the hubitat / smartthings side to give a percentage.

ebaauw, 2019-02-28:
> if battery percentage could be added

The REST API plugin should expose `config.battery` for each `/sensors` resource from the _Power Configuration_ cluster.

> Battery percentage is reported from 0 - 200

That's the ZCL spec, battery percentage in units of 0.5%.

> it has an optional charging circuit so you can attach an 18650 cell for battery backup capability

I figured that from the links.  It should report _Power Source_ (0x0007 in the _Basic_ cluster) as "Mains (single phase) with battery backup", or does it change the attribute when the battery is connected?

martynwendon, 2019-02-28:
Yeah that was one of the reasons the chap that creates them added the battery feature as it mitigates against battery end devices dropping off the ZigBee network.  Very useful for Xiaomi stuff as you are no doubt aware is leave/rejoin challenged :-)

The Power Source attribute doesn't change unfortunately, but that's definitely something I'll recommend for the future - it would also be good to have a power fail event of some description.

I've seen your checkin for the whitelisting, I guess my next question would be is there a way to do that now on my live system, or do I have to wait for the next release and update?

ebaauw, 2019-02-28:
> it would also be good to have a power fail event of some description

Been looking for that event for my Heiman Siren.  I think/hope it reports this info through the _Power Descriptor_, which the REST API plugin doesn't read, and the GUI doesn't seem to display in full.  Heuristically, when the _Battery Percentage_ (in the _Power Configuration_ cluster) drops, it's on battery power, and when the _Battery Voltage_ increases, it's mains powered again.  The _Battery Voltage_ isn't exposed thru the REST API...

Of course, when it's a power failure (rather than just a blown fuse or disconnected plug), the UPS solution will signal that, as the wired Ethernet connection fails.

> I guess my next question would be is there a way to do that now on my live system, or do I have to wait for the next release and update?

If you're running deCONZ v2.05.59 on a Raspberry Pi, you could compile the REST API plugin yourself, from my commit. See https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional.  That would still involve restarting deCONZ (I find it crashes when you replace the plugin when it's running).

martynwendon, 2019-02-28:
Thanks again, I cloned your branch and built it, stopped deconz and copied across the new plugin file.

Must be missing something though as I can't seem to get anything new appearing in the rest /sensors output?

I even deleted, reset and re-paired the sensors (I have three of them).

I opened the network for joining from both the old web UI and the new phoscon UI, but still nothing.

Have I missed at step?

ebaauw, 2019-02-28:
Odd.  Did you move the original plugin out of the `/usr/share/deCONZ/plugins` directory?  Just renaming it won't do it - deCONZ loads all files in that directory, regardless of the suffix.

Opening the network from the old web app, or searching for devices from Phoscon should do the trick.  The sensor is mains powered, so there shouldn't be any issues to read it.  You could try reading the attributes from the _Basic_ cluster in the GUI (while the network is still open/search is still active).

martynwendon, 2019-02-28:
>  You could try reading the attributes from the _Basic_ cluster in the GUI (while the network is still open/search is still active).

This sorted it!  Easy when you know how :-)

config.battery showed up as null originally, but a read on 0x0021 attribute in the GUI sorted that.  Will set up some reporting on the various attributes next.

Thanks again for all your help, really pleased that these were pretty much supported out-of-the-box!

martynwendon, 2019-03-01:
Hey @ebaauw I have a further question, which may just be my misunderstanding of how deCONZ works!

I've got three of these sensors on deCONZ now and am getting swamped with events from them, despite me setting appropriate levels of attribute reporting.

With my ZigBee Sniffer I can see that deCONZ is spamming the heck out of the devices, sending Read Attribute requests for the Illuminance Measurement Cluster.  

I'm sure I've got these setup right - as far as I can see setting attribute reporting is for the autonomous reporting from devices and doesn't have anything to do with any "polling".

Have I done something wrong?  Or is this expected behaviour?

ebaauw, 2019-03-01:
It's a mains-powered device, so deCONZ might be eager to poll it.  However, it shouldn't do so, when the value has been updated recently enough (which should be the case with attribute reporting).  The Hue motion sensor is configured to send periodic reports every 5 minutes for motion, light level, and temperature; and every 2 hours for battery percentage.  Do you see _Report Attributes_ messages for the _Illuminance Measurement_ cluster in the sniffer?

martynwendon, 2019-03-01:
Yes each sensor responds immediately with a Report Attributes Response.

I've set attribute reporting the same way as I had them under SmartThings / Hubitat so min/max/change like so:

Temperature - 10/300/20
Humidity - 60/300/100
Illuminance - 5/300/2000
Pressure - 60/300/1
Battery - 60/300/2

deCONZ only seems to be polling for Illuminance, not the others.

I've attached a screenshot of the sniffer, the sensors are 75F0, 5DFA and F70B.
[image]

ebaauw, 2019-03-01:
Is the binding from the _Illuminance Measurement_ cluster to the gateway active?  Do you see _Report Attribute_ commands from the sensors to the gateway?

martynwendon, 2019-03-01:
Ah I see what you were asking, apologies that I misunderstood.

No there weren't any autonomous Report Attribute commands from the sensors, it's just the polling (Read Attribute / Read Attribute Response) from deCONZ.

I couldn't see a way to check what cluster bindings are set, is there a way to view them?

This may have caught me out as under Hubitat / SmartThings if you call configureReporting() in your driver and a binding hasn't already been set, it creates one automagically - I've now learnt that this is a two-step process under deCONZ :-)

In any case, I've gone through each sensor now and manually set a binding from each cluster to deCONZ and the polling has stopped.

While I created my own mess so to speak, I suspect that this could be considered a bug given the rate that the polling was occurring?

After all, the sensor *was* responding to every poll, which I'd imagine should constitute the value being updated recently enough to prevent a subsequent poll?  I can't imagine that polling until an autonomous Report Attribute is received is desired behaviour?  

Thanks again for the quick reply and help!

ebaauw, 2019-03-01:
> I couldn't see a way to check what cluster bindings are set, is there a way to view them?

No.  I usually try an _Unbind_ from the GUI: if that was successful, I know the binding was there.  Afaik it doesn't hurt to try multiple binding requests.  While you're running the sniffer, you can check whether the sensor sends a response.  I think the request is ZDP cluster 0x0021 and the response 0x8021.

> I've now learnt that this is a two-step process under deCONZ :-)

Indeed.

closed 2019-07-06 by stale[bot]