dresden-elektronik/deconz-rest-plugin #1309: Plugs not controlable after restart unless i control it manually first

sarakha63, 2019-03-01:
Hello

I have a quite annoying issue with deconz and the rest api. I have multiple sensors and some lights and plugs.  And everything works flawlessly but once i restart deconz (used on a debian system and Home Automation Solution Jeedom) i loose all controls on my plugs (for example xiaomi smart plug) every action gives me

Erreur lors de la requete : 127.0.0.1:8484/api/901D065383/lights/00:15:8d:00:01:27:aa:35-01/state(PUT) erreur : 3 => resource, /lights/00:15:8d:00:01:27:aa:35-01, not available

Stating that the endpoint cluster is not available. The only solution for it to work is to first make the plug send something (manual action) then i guess the api rebuilds some infos... and directly after i regain control over rest api.

This is quite annoying for a Home Automation System. We are actually working hard on adding deconz support on Jeedom (popular french Home Automation Solution), everything is going great for now but this issue seems annoying. We have some other minor issues (will discuss about it later)

ebaauw, 2019-03-01:
What deCONZ version are you on?  Might be #1207, which is fixed in v2.05.59.

sarakha63, 2019-03-01:
We are currently using 02.05.59

ebaauw, 2019-03-01:
Hm, I do find that devices might not seem to react shortly after (re)starting deCONZ, but I don't get any resource not available errors.  I don't have any Xiaomi plugs, though, but my innr plugs work as soon as deCONZ has discovered the route to them.

sarakha63, 2019-03-01:
I just tried, even lots of minutes after starting i still have the issue. Could'nt Deconz Fallback to last Known infos after restart.
I understand that deconz wants to interview the network after restart. But making commands not being sent until then is i think a major issue. As in this case it's not the device not reacting but deconz not sending at all

ebaauw, 2019-03-01:
> I understand that deconz wants to interview the network after restart. But making commands not being sent until then is i think a major issue. As in this case it's not the device not reacting but deconz not sending at all.

deCONZ can only send a unicast message to a device when it knows how to route this message across the Zigbee mesh network.  It should do a just-in-time route discovery, but this might get lost, since there's a lot of traffic after (re)starting, querying all the nodes.  This should be resolved within a minute or two after startup.  
Note that routes are asymmetrical: the device still knows how to reach the coordinator/gateway (since the device hasn't restarted), but the gateway doesn't know how to reach the device.  Typically, the device will react to broadcast/groupcast commands from the gateway.

> Could'nt Deconz Fallback to last Known infos after restart.

I'm not sure.  Afaik, the coordinator's neighbour table is stored in the RaspBee or ConBee by the firmware, so that should survive a restart of deCONZ.  The routing table (how to reach nodes further away), however, is stored in memory in the deCONZ core.  I don't think this information is persisted in the database (which is maintained by the REST API plugin), so it needs to be rebuilt after a restart.

I'm not sure if the "resource not available" error is actually related to the route discovery.  It smells like the REST API plugin doesn't correctly/completely restore the resource from the database on startup, requiring a message from the device to make it available.  I think I had a similar issue for my (end-device) `lumi.ctrl_neutral` switches a couple of versions ago, but I no longer get any "resource not available" errors for these.  I'm on v2.05.59 as well.

sarakha63, 2019-03-01:
Ok so we have to dig a little i guess. I totally get the first point, the second one i wonder if persisting some infos could be usefull (future will surely bring some strange devices i guess as zigbee is clearly getting more popular).

your last point is interesting, it means that just correcting the not available issue would maybe be enough.

But its clearly that upon a restart, i cant control the device until it (the device) sent something which may take one hour (heartbeat) or needs a manual intervention

sarakha63, 2019-03-01:
One more precision after a restart if i ask for sensors. the device is here but only with ZHAPower (consumptions infos)  only Smart Plug infos and actions is missing

sarakha63, 2019-03-02:
And one more precision. After a restart the device also disappears from phoscon and is no more visible on lights menu until i manually trig the plug

ebaauw, 2019-03-02:
Phoscon uses the REST API to talk to deCONZ, so when the plug's `/lights` resource is missing, Phoscon won't show it.

closed 2019-07-07 by stale[bot]