dresden-elektronik/deconz-rest-plugin #1310: Lqi and network datas in rest api

sarakha63, 2019-03-01:
Hello

We are creating a wide integration of deconz on a Home Automation Software. And we really would like to give the ability to our users to see a kind of network graph. Unfortunately it seems that we have no way of getting Lqi and links informations via Api.

Is there a way of getting this kinf of infos ? Or is it planned? 

Regards

ebaauw, 2019-03-01:
See #219.

sarakha63, 2019-03-01:
Hello  thanks for the answer we are going to play with it. Eventough as i read on the issue it seems we must consider this unreliable

sarakha63, 2019-03-01:
We just tried.

I understand that only routers can give the neighbours and linq quality. that is ok and makes sense.

But in that case i guess we are missing the controler neighbours to make a network graph...

ebaauw, 2019-03-01:
Yes, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/219#issuecomment-468632570.

closed 2019-07-06 by stale[bot]