dresden-elektronik/deconz-rest-plugin #1312: Light is hanging on low level = 1 instead of 0

Ronoon, 2019-03-01:
Got a problem with the lights that will sometimes not get off. 
So after sometimes finding the lights on instead of off,  this time i catched the light on in the phoscon app and it showed that the light level was 1 instead of 0
Seems like the loop should be triggered by 1 instead of zero to confirm that the light is defined of cause the zero seems to be not every time to triggered (happens to me 1 to 3 times a day).
This seems to explain why the lights run from low level up to max level if you trigger the motion sensor a second time when the lights don´t turn off.

manup, 2019-03-02:
A few questions:
* Which kind of light is it?
* How do you turn off the light?

In Zigbee Light Link there is not exactly a 0 light level, 1 is the minimum light level. 0 has a special meaning for certain commands, but this is better described in the ZLL/ZCL spec.

Ronoon, 2019-03-02:
It is a INNR GU10 light
[image]
[image]

[image]
it is turned on/off by the phoscon app like shown in the picture
and with the off command in the picture it is hanging like this:
[image]

Seems not to reach the 0 position so i have to set it there manually to 0 to shut it down

manup, 2019-03-02:
Does it turn off when you click on the light bulb icon (or the light group icon)?

I have only the E27 innr light for testing but this one works.

Ronoon, 2019-03-02:
yes that is working 
i think there is something with the ramping going on that i cannot see
if the light is on and i go through the sensor the light will level up again an then the timing is working its way again until the light goes off 
got that in every version so far and also with different configurations of the lights.
was changing the light group to two different groups an got the same result
also changed the bulbs to different ones but the result again was the same

closed 2019-07-07 by stale[bot]