dresden-elektronik/deconz-rest-plugin #1314: Pairing Hue bulbs

hexolite, 2019-03-02:
Hi, 
I am new to deCONZ community and I am trying to add my Hue Bulbs to deCONZ but I can't... 

I want to use Touchlink but how can I do that with Phoscon Web App? 

Thanks in advance

Ronoon, 2019-03-02:
Easiest way is to reset the hue bulbs and scan them with the phoscon app.
You need the Phillips hue smart dimmer to reset the bulbs.
This is how it worked for my hue bulb without any problems

hexolite, 2019-03-02:
Hi! Thanks for your reply! The problem is that I don't have the dimmer... if I can I'd prefer to avoid buying one... 
I think there is another with touchlink and old web app (not phoscon) but I can't find how to do it... 

If I buy the dimmer can you be more precise about how to reset bulb and make them detectable ? Thanks !

Ronoon, 2019-03-02:
Here is the german reset with the Hue dimmer.  I have put a Translation to the german text.
Hope this will be helpful.

Philips Hue-Leuchte per Fernbedienung zurücksetzen
Philips Hue Bulb to reset via Hue Dimmer
Das Zurücksetzen einer Philips Hue-Lampe via Fernbedienung erfordert je Gerät nur einen Schritt:
Hue Dimming Switch: Ein- und Aus-Taste für fünf Sekunden halten.
Hue Dimming switch: press the on and the off switch 5 seconds
Die Philips Hue-Leuchte setzt sich nun auf ihre Werkseinstellungen zurück.
The light will go back to manufactury defaults
Die Lampe sollte dabei kurz aufleuchten. 
the light will give a blink sign during the reset
Falls dies nicht funktioniert, die Fernbedienung näher an die Leuchte halten.
If that is not happening,  hold the dimmer near the bulb

Smanar, 2019-03-03:
To use the touchlink (But I haven't used it myself)
Help > Open Wireless Light Control (2016) to have the old webapp
then menu / setting /reset device via touchlink.

But your bulb needd to be very close to the gateway and perhaps you need to power off/on the bulb before.

hexolite, 2019-03-05:
Hi Smanar, 
Thanks for your reply! It worked great 👍

closed 2019-07-10 by stale[bot]