dresden-elektronik/deconz-rest-plugin #1316: Door/Window sensors and group cluster

stevenwfoley, 2019-03-03:
Does anyone have any experience with a door/window contact sensor that supports the Groups cluster? I'd like to have my door sensors operate if the phoscon gateway is down.

stevenwfoley, 2019-04-26:
I bought an Xiaomi Aqara MCCGQ11LM window/door sensor, and it seems to support the Groups cluster. I think this because there is an item named "Groups" in the Cluster Info dialog for the device in deconz. Does anyone know how to take advantage of or configure this groups cluster with a sensor?

manup, 2019-04-26:
The Groups cluster is a client side (grey color), usually this is used so that the sensor could control group membership of other devices, like put a light to a specific group.

[image]

I can only image the only way to get the sensor work when the gateway is offline is a binding from the On/Off cluster to a group. 

[image]

Unfortunately this doesn't seem to work, the binding can be created but sensor still just reports to the gateway. Also tried with unicast binding to a IKEA light but still the sensor only reports to the gateway.

stevenwfoley, 2019-04-30:
Thanks for trying manup.

closed 2019-04-30 by stevenwfoley