dresden-elektronik/deconz-rest-plugin #1320: Rediscover Hue after power out

Gregsen, 2019-03-03:
Sometimes, I turn my lights off (as in: using the switch on the wall). When I turn it on again, deConz has lost its connection to the light. I have to rescan for the light, then it is working again. 

Is there a way to to that automatically? Scanning periodically for the devices?

**Light:**
 Phillips LWB010 
Version: 1.23.0_r20156
**deConz**
Version 2.05.58 / 30.1.2019
Firmware 262F0500

Oliviakrkk, 2019-03-03:
Have you tried waiting? Mine come back within 30mins~hour

ebaauw, 2019-03-03:
They should come back almost instantly, as the light broadcasts a device announcement on startup. which deCONZ picks up.  Of course, if the light is out of direct reach of the RaspBee/ConBee, and all lights in between are powered down as well, this won't help you.  Morale: don't use the wall switches and leave the lights powered on all times.

Gregsen, 2019-03-03:
@Oliviakrkk Unfortunately, waiting did not help. Even after 3 hours, it has not found the light.

@ebaauw It's hard to overcome decades of muscle memory :smile: . I am just wondering, because, if I scan for new devices, it does find the light again. So the distance should be fine, right? It is just not finding it on it's own when I turn it on. So I thought maybe there is an API way to rescan (something like check via cron for the lights and if its not reachable, scan for lights)

ebaauw, 2019-03-03:
> maybe there is an API way to rescan

You can initiate a scan by a POST on `/lights` or `/sensors`.  To see the results, do a GET on `/lights/new` or `/sensors/new`.

closed 2019-07-08 by stale[bot]